MarlinFirmware/Marlin #19786: No serial communication on port 3 - BTT SKR1.4

Drzet, 2020-10-17:
### Bug Description
ESP8266 in dedicated WiFi port does not work since 2.0.6
This issue has been addressed many times before. It required modifications to Hardwareserial.h in PlatformIO to properly define UART3 pins. Latest update with the new `BOARD_CUSTOM_BUILD_FLAGS -DLPC_PINCFG_UART3_P4_28`  definition was supposed to fix it.
However, since 2.0.7 UART 3 has stopped working. Reverting HardwareSerial definitions to 'old-style' does not fix the problem either.
It does not seem to be caused by the definitions any more. M43 confirms correct pin assignment:
Recv: PIN: 4.28        TXD3                                   Output = 1
Recv: PIN: 4.29        RXD3                                   Input  = 1

Tested with 2 ESP boards. Reverting to 2.0.6 fixes the issue

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5396389/config.zip)

UART3 defined as LPC_PINCFG_UART3_P4_28

**Expected behavior:** [What you expect to happen]
Serial connection between ESP8266 and SKR
**Actual behavior:** [What actually happens]
WiFi connects and web interface loads but serial connection can't be established
#### Additional Information

ellensp, 2020-11-02:
Just tested this on current bugfix
Brand new skr 1.4 (turbo) and esp-01s
In Configuration.h I only changed Motherboard BOARD_BTT_SKR_V1_4_TURBO, serial_port 3, and serial_port_2 -1,
and in platformio.ini default_envs = LPC1769
You do need to match the BAUDRATE in marlin (default is 250000) with the baud rate in esp3d (default is not 25000) 
Check you ESP3D Settings

ellensp, 2020-11-02:
Just ported your configs over to bugfix.
Only change are that motherboard is set to the turbo and TEMP_SENSOR_0 and TEMP_SENSOR_1 where set to 1.
esp3d is still working fine.
I cannot find any issue.

boelle, 2020-11-02:
i just asked the bot to close this one if nothing happens within 5 days
i do so on the ground that there are no issue to be found

if there are an issue it should be explained in better detail or with a video

Drzet, 2020-11-02:
Matching baud rate between the printer and ESP solves the problem. Thank you

closed 2020-11-02 by Drzet