MarlinFirmware/Marlin #19787: No msteps on TMC 5160 + SKR pro v1.1

CaezarCZ, 2020-10-17:
### Bug Description

Sorry for the noob question or problem, but i changed my marlin to 2.0.7.1 (i tried also 2.07.2) and worked perfectly until power off.  After power off no movement on G28, no movement after reflasing. When i try the M122 (lower in text) i get diferrent setup than in marlin ( #define DEFAULT_AXIS_STEPS_PER_UNIT   { 784, 769, 3231, 2760 } )  And i dont have a idea where can be a problembut the tstep and the PWM scale have the same numbers.

I also tried the on the start M17 XYZE without succes 
The drivers are from BTT as same as the board 

X	Y	Z	Z2	E
Enabled		false	false	false	false	false
Set current	630	630	630	630	650
RMS current	622	622	622	622	641
MAX current	877	877	877	877	904
Run current	12/31	12/31	12/31	12/31	12/31
Hold current	6/31	6/31	6/31	6/31	6/31
Global scaler	129/256	129/256	129/256	129/256	133/256
CS actual	31/31	31/31	31/31	31/31	31/31
PWM scale	4294967295	4294967295	4294967295	4294967295	4294967295
stealthChop	true	true	true	true	true
msteps		0	0	0	0	0
tstep		4294967295	4294967295	4294967295	4294967295	4294967295
PWM thresh.
[mm/s]
OT prewarn	true	true	true	true	true
triggered
 OTP		false	false	false	false	false
off time	15	15	15	15	15
blank time	54	54	54	54	54
hysteresis
 -end		12	12	12	12	12
 -start		8	8	8	8	8
Stallguard thrs	0	0	0	0	0
uStep count	65535	65535	65535	65535	65535
DRVSTATUS	X	Y	Z	Z2	E
sg_result	1023	1023	1023	1023	1023
stallguard	*	*	*	*	*
fsactive	*	*	*	*	*
stst
olb		*	*	*	*	*
ola		*	*	*	*	*
s2gb		*	*	*	*	*
s2ga		*	*	*	*	*
otpw		*	*	*	*	*
ot		*	*	*	*	*
Driver registers:
		X	0xFF:FF:FF:FF	 Bad response!
		Y	0xFF:FF:FF:FF	 Bad response!
		Z	0xFF:FF:FF:FF	 Bad response!
		Z2	0xFF:FF:FF:FF	 Bad response!
		E	0xFF:FF:FF:FF	 Bad response!
Testing X connection... Error: All HIGH
Testing Y connection... Error: All HIGH
Testing Z connection... Error: All HIGH
Testing Z2 connection... Error: All HIGH
Testing E connection... Error: All HIGH

### My Configurations
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5396496/configuration.zip)

**Expected behavior:** motor movement

**Actual behavior:** no motion, strange TMC debug data

thisiskeithb, 2020-10-18:
Make sure you're initializing your EEPROM after each flash by sending `M502` followed by `M500`.

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

CaezarCZ, 2020-10-18:
I tried both without the succes

ellensp, 2020-10-18:
Its a TMC, did you have you external power supply on before the main controller, as TMC's need the external power to be configured

CaezarCZ, 2020-10-18:
I have tested even the usb power (with external power source) as same as the full power from the external source , fuses are OK. I tried aslo diferent version of TMC library, different setting of spreadcycle, software spi etc. So i think that shoud be some deeper problem like https://github.com/MarlinFirmware/Marlin/issues/15055 , which i cannot recognize with my basic knownledge , also i will have time for testing probably by the end of next week.

CaezarCZ, 2020-10-24:
I also tried newest bug fix i think that last funcional version was 2.0.5.4,  after that i had a problem with motion control and iam not sure what changed this

CaezarCZ, 2020-10-30:
the problem was probably with the alowed software spi / SOFTWARE_DRIVER_ENABLE combination

closed 2020-10-30 by CaezarCZ