MarlinFirmware/Marlin #19795: [BUG] UBL with PROBE_MANUALLY doesn't lower to MANUAL_PROBE_START_Z and stays at Z_CLEARANCE_BETWEEN_PROBES

qurvento, 2020-10-18:
### Bug Description

UBL with defined PROBE_MANUALLY doesn't move to MANUAL_PROBE_START_Z and stays at Z_CLEARANCE_BETWEEN_PROBES height when manually building a mesh via LCD.

### My Configurations

Creality Ender-3 v4.2.2
Marlin 2.0.7.1
AUTO_BED_LEVELING_UBL, PROBE_MANUALLY, LCD_BED_LEVELING
[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/5397588/cfg.zip)

### Steps to Reproduce

1. Select "Motion/Unified Bed Leveling/UBL Tools/Manually Build Mesh".

**Expected behavior:** moves to center and then to Z = MANUAL_PROBE_START_Z

**Actual behavior:** moves to center and stays at Z = Z_CLEARANCE_BETWEEN_PROBES

boelle, 2020-10-20:
is this related to #19814 ?

qurvento, 2020-10-24:
no

sjasonsmith, 2020-11-16:
I have posted a PR to resolve this. I don't know yet whether it will be accepted and merged. Please feel free to test it and comment on your results in the PR.

boelle, 2020-11-17:
> Thank you.

did you test it?

sjasonsmith, 2020-11-19:
This change was merged into the bugfix-2.0.x branch today. I'm closing this issue out, go ahead and reply if you encounter problems using the change.

closed 2020-11-19 by sjasonsmith