MarlinFirmware/Marlin #19799: [BUG] CALIBRATION_GCODE/BACKLASH_COMPENSATION - G425 Odd Behavior

Protean-Man, 2020-10-18:
Configs & Log File:
[repetier.log](https://github.com/MarlinFirmware/Marlin/files/5397905/repetier.log)
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5397910/Configs.zip)

I am trying to implement auto calibration of offsets and backlash compensation and have only found problems. 
When calling G425 on a multi-tool printer the calibration sequence starts as expected but once the height calibration is finished it seems to move back to the first tool storage slot - except the X coordinates are wrong and it ends up smashing into the tool storage.
I assumed this was an attempt to change tools and test the Z height of other tools but when calling G425 T0 - the exact same thing happens. 
There are no measurements being taken of the sides of the nozzle, only the height against the calibration probe.

**Expected Behavior:** 
1 - Call G425
2 - printer moves T0 to calibration probe and tests top and sides (3 times each?)
3 - printer then goes to next tool and repeats for all tools
4 - Offsets and backlash is applied and saved to EEPROM

**Actual Behavior:**
1 - Call G425
2 - printer moves T0 to calibration probe and tests top 3 times
3 - carriage then moves to opposite end of printer where tools are stored and crashes into tool storage - so I do a hard stop on the printer. 

CALIBRATION_OBJECT_CENTER     { -31.0, 7.0,  2.3 } - **This is at the front of the printer at the edge of the bed.**
Tool Storage are at varying X positions and all at Y374
Why would Calibration move from the front of the printer where the Calibration Probe is all the way to the back of the printer on the Y plane - during calibration? The X doesn't seem to change.

In the Log file it doesn't even mention the Y move after Calibrating the Z height on T0:
<<< do_blocking_move_to  X-31.00 Y7.00 Z3.69
> 10:40:03.336 : >>> do_blocking_move_to  X-31.00 Y0.00 Z3.69
**> 10:40:03.336 : >  X-31.00 Y0.00 Z3.69** _last calibration of Z but the Y should be 7 and not 0. Is this it?_
> 10:40:03.838 : T:199.86 /200.00 B:59.91 /60.00 T0:199.86 /200.00 T1:66.67 /0.00 @:27 B@:47 @0:27 @1:0
> 10:40:03.913 : echo:busy: processing
> 10:40:04.837 : T:199.89 /200.00 B:60.01 /60.00 T0:199.89 /200.00 T1:66.36 /0.00 @:27 B@:37 @0:27 @1:0
**< 10:40:05.583 : N25 M112*22** _me calling emergency stop because it is about to smash into the back after moving from the front where the calibration probe is_
< 10:40:06.740 : Printer reset detected - initializing

shitcreek, 2020-10-21:
Is this is same issue: https://github.com/MarlinFirmware/Marlin/issues/19712
What does `G425 V` report?
Are you certain your true calibration object's location is within 5mm and the probe can reach all four sides?

Protean-Man, 2020-10-21:
My X&Y endstops are at -52 & -34. There is ample space around my calibration sensor. The sensor is at the front of the printer. 
What is weird is what happens after it runs through the Z calibration (senses Z 3 times), where it then moves to the very back of the printer and smashes into the tool storage. 

what do you mean "true calibration object"?
The location is set under CALIBRATION_GCODE

boelle, 2020-10-22:
> Is this is same issue: #19712
> What does `G425 V` report?

Protean-Man, 2020-10-23:
@boelle no but they may be related? 
It was different behavior that I am reporting here but it is still related to axis limits or possibly something else? - I am unsure.
#19712 is moving past X min and this issue is the carriage moving beyond Y Maximum.

Outside of the axis limit ignoring, the other issue is that after the Z calibration probing it does not run the X and Y calibration probing - instead the carriage just moves away and beyond axis limits.
Hope this helps.

closed 2020-12-15 by github-actions[bot]