MarlinFirmware/Marlin #19800: [FR] Neopixel support on STM32F1

Loreagle, 2020-10-18:
I discovered that Neopixels works perfectly on my MKS Robin Nano 1.2 using this library: 

Adafruit NeoPixel=https://github.com/ccccmagicboy/Adafruit_NeoPixel/archive/master.zip

It would be nice to add it to lib_deps at least for Robin Nano and also add LED_PIN in pins file

fiechr, 2020-10-25:
I would really appreciate NeoPixel support for STM32F1, too.

I've tried the lib, but only get this, when building for my BTT Mini E3 v2.0:

    Building in release mode
    *** [/home/cf/.platformio/packages/framework-arduinoststm32-maple/STM32F1/variants/generic_stm32f103r/board.cpp] 
    /home/cf/src/Marlin2x_cf/.pio/build/STM32F103RC_btt_512K/.sconsign38.dblite: No such file or directory
    ==================================================== [FAILED] Took 6.20 seconds ====================================================

Loreagle, 2020-10-25:
> I would really appreciate NeoPixel support for STM32F1, too.
>
> I've tried the lib, but only get this, when building for my BTT Mini E3 v2.0:
>
> ```
> Building in release mode
> *** [...]

Did you add the library to lib_deps?

fiechr, 2020-10-25:
Yes, at least I think so.

I replaced the default 1.5.0 lib with the URl from above:

```
#NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0
NEOPIXEL_LED            = https://github.com/ccccmagicboy/Adafruit_NeoPixel/archive/master.zip
src_filter=+<src/feature/leds/neopixel.cpp>
```

Loreagle, 2020-10-25:
Try to add this under your board in platformio.

`lib_deps        = ${common.lib_deps}
  Adafruit NeoPixel=https://github.com/ccccmagicboy/Adafruit_NeoPixel/archive/master.zip`

sammy2142, 2020-10-25:
Aren't you supposed to enter the location of Delay.h into the Adafruit_NeoPixel.cpp file as well?
https://github.com/ccccmagicboy/Adafruit_NeoPixel/commit/087d9f714e104426029b7cbbddfd8bae7371710d

Ps. the ccccmagicboy fork just appears to be a clone of CommandoreBombardiero work: https://github.com/CommandoreBombardiero/Adafruit_NeoPixel

Loreagle, 2020-10-25:
> Aren't you supposed to enter the location of Delay.h into the Adafruit_NeoPixel.cpp file as well?
> [ccccmagicboy/Adafruit_NeoPixel@087d9f7](https://github.com/ccccmagicboy/Adafruit_NeoPixel/commit/087d9f714e104426029b7cbbddfd8bae7371710d)
> 
> Ps. the ccccmagicboy fork just appears to be a clone of CommandoreBombardiero work: https://github.com/CommandoreBombardiero/Adafruit_NeoPixel

Yes you need to put the location of Delay.h but I hope there is a way to do so automatically if this library will be implemented

fiechr, 2020-10-25:
Ok, I'm getting so far, that it complains about the Delay.h now, but since I don't even know where the Delay.h actually is, this is where I fold for now.
But I'm still happy to test if someone creates a pull request or similar.

ellensp, 2020-10-25:
change the path for delay to
#include "../../../src/HAL/shared/Delay.h"

ellensp, 2020-10-25:
@Loreagle  unfortunately this method only works for some..   there are other factors in play. Its not the full answer, its 1/2 an answer at best

Loreagle, 2020-10-25:
> @Loreagle  unfortunately this method only works for some..   there are other factors in play. Its not the full answer, its 1/2 an answer at best 

If I'm not wrong I tried with your library before and It was working only sometimes, when recompiling colors were messed up.
With this library, at least on my robin nano 1.2, I tried recompiling a lot of times and always worked.

sjasonsmith, 2020-10-25:
The Marlin delay function is currently off by about 50% for these STM32F1 boards, which throws off timing. It’s probably on the edge of working properly, so behaved erratically.

There is an open PR to improve delay timing, but I have admittedly been holding it up on my own testing.

Sebazzz, 2020-12-16:
I got it *almost* working - you can check the required repositories here:
- https://github.com/CR6Community/Marlin/tree/experiment/extui-btt-skr-tft-neopixel
- https://github.com/CR6Community/Adafruit_NeoPixel

The only thing that does not work is turning the neopixel off: One green LED stays on. All other colors seem pretty stable.

thisiskeithb, 2020-12-16:
> I got it almost working - you can check the required repositories here:

Does it work on all `STM32F1` boards or just those used in the CR6SE? The main issue (and why it's disabled in the main Marlin repo) is that NeoPixels works/mostly works on some `STM32F1` boards & not others.

Sebazzz, 2020-12-16:
> Does it work on all `STM32F1` boards or just those used in the CR6SE?

Let's just say on the BTT SKR CR6 board. My remark, just like the ones before me, is for the tinkerer which wants to play on his own with the neopixels.

That said, a worthy alternative might be the [FastLED](https://learn.adafruit.com/adafruit-neopixel-uberguide/advanced-coding) library which does appear to support STM32 at first glance (but it needs more research).

X-Ryl669, 2021-01-06:
There's also https://github.com/rogerclarkmelbourne/WS2812B_STM32_Libmaple that makes use of the SPI's DMA engine on STM32 to generate the right timing (so it's not dependent on a correct DELAY_NS) function.

X-Ryl669, 2021-01-21:
I propose to implement a FastLED path since IMHO NeoPixel is a dead end. It's too heavy on performance (it burns CPU for generating timing which is a no-no for realtime system like Marlin) and breaks systick period on some platforms.

Sebazzz, 2021-01-24:
> 
> 
> There's also https://github.com/rogerclarkmelbourne/WS2812B_STM32_Libmaple that makes use of the SPI's DMA engine on STM32 to generate the right timing (so it's not dependent on a correct DELAY_NS) function. It _should_ be compatible with NeoPixel API

Yes, but that doesn't work unfortunately for boards that actually use SPI for other things, like SD card communication.

X-Ryl669, 2021-01-24:
Yes, you are right. I've just started studying the mess of handling WS2812 on STM32. So far, I'm leading to these conclusions:
1. NeoPixel is a brute force approach and using this means that changing the light output (color or brightness) implies eating CPU time from printing. In addition STM32 does not have a correct delay method so the control of the leds is somehow failing to work as expected. So Marlin is conservative here, and it's only providing optional feature here (that is, change the led color when the bed is heating up and no printing is happening or set the printer's light once for all upon starting printing) or for a specific G-Code command (meaning it's pausing printing when this command is parsed/executed).
2. FastLED, while the name of the project implies that it's better than the former, I've found while reading the code that it's calling itself "fast" because they rewrote the main delay logic in assembler. It's still bit banging underneath. It imports so many dependencies that are almost duplicated from Marlin that it does not worth the engineering time for porting it in Marlin either.
3. There's also [FAB_LED](https://github.com/sonyhome/FAB_LED) which is doing what FastLED is doing within 1/5 of the Flash budget and 1/3 of the SRAM budget (and 1/10 the code size). All in all, it'd be better to use the latter than the former.
4. On STM32, the SPI hack solution would mess up with other SPI port, since it changes the SPI's frequency.
5. On STM32, there's also the possibility to program a Timer and use DMA to generate the signals for the LEDs (that's the approach used [here](https://github.com/Crazy-Geeks/STM32-WS2812B-DMA). This means almost no CPU used for generating the signals. I've quickly checked the code and I think it's possible to implement this on Marlin. It forces using some specific pins but that's something that I think it's acceptable. It's a bit like what Octo2811 is doing but for STM32 instead of Teensy.

Sebazzz, 2021-01-24:
For STM32 I'd say that flash and RAM budget is not very important. I have a RET6 processor and *plenty* of flash space and free RAM. 

But it seems that Neopixel implementation would need to be delegate to the HAL possibly. Good digging you did @X-Ryl669!

sjasonsmith, 2021-01-24:
I found one approach that used a timer to trigger DMA to GPIO registers. In this way you were not coupled to specific pins on the board, and you could send unique patterns to many strands with a single timer, as long as they are all in the same GPIO port.

I was going my to try it, but had too many conflicts for my time.

X-Ryl669, 2021-01-24:
Do you have some documentation or link about your tests ? 
Is it in one of your branches ? 
Was it described from [this page](http://www.martinhubacek.cz/arm/improved-stm32-ws2812b-library) ?

sjasonsmith, 2021-01-24:
Yes, that seems to be it. I now see that it takes three DMA channels. I suppose that will be the next scarce resource to fight over.

X-Ryl669, 2021-02-04:
With #20901 and @ellensp's fork, it's working perfectly on WS2812b strip and STM32F1:

[image]
[image]
[image]
[image]
[image]

X-Ryl669, 2021-02-06:
@Loreagle The bugfix branch now include the fix. Please test on your side, and if it's ok for you, close the issue. Thanks!

Loreagle, 2021-02-06:
> @Loreagle The bugfix branch now include the fix. Please test on your side, and if it's ok for you, close the issue. Thanks!

Unfortunately I'm using klipper now and I can't test it but if it's working for you I think is okay since I have a sapphire pro with robin nano too 😅

Sebazzz, 2021-02-06:
I will test it on my BTT SKR CR-6 board.

Met vriendelijke groet,
Sebastiaan Dammann

Sebazzz, 2021-02-06:
@X-Ryl669 Yes, it works perfectly!

Loreagle, 2021-02-06:
Perfect I can close this so! Thanks :)

Sebazzz, 2021-02-07:
> 
> 
> Perfect I can close this so! Thanks :)

Can you? I think the warning in the sanity check still needs to be removed.

thisiskeithb, 2021-02-07:
> > Perfect I can close this so! Thanks :)
> 
> Can you? I think the warning in the sanity check still needs to be removed.

Has this been fully tested on more than one `STM32F1` board?

I'd like to see it tested on more boards before we declare it working on all since we already get enough support requests about NeoPixels on `STM32F1` (even with the sanity check) as it is.

thisiskeithb, 2021-02-07:
Reopening for further testing/discussion.

X-Ryl669, 2021-02-07:
To collect working board, mine: MKS Robin Nano V1.2 (the mks_robin_nano35) is working.

Adi146, 2021-02-08:
@X-Ryl669 which Neopixel library do you use? I get this error _.pio\libdeps\mks_robin_nano35\Adafruit NeoPixel/Adafruit_NeoPixel.h:361:3: error: 'GPIO_TypeDef' does not name a type_ with the default Adafruit NeoPixel@1.5.0 lib

X-Ryl669, 2021-02-08:
I'm using @ellensp's fork version, see [here](https://github.com/MarlinFirmware/Marlin/issues/19800#issuecomment-716145359)

Sebazzz, 2021-02-08:
Yes, under the particular `[build_configuration_name]` you are using, redirect the Adafruit Neopixel library like this:

```ini
custom_marlin.NEOPIXEL_LED = Adafruit NeoPixel=https://github.com/ellensp/Adafruit_NeoPixel
```

Floppy, 2021-02-11:
Confirmed working on a BTT SKR Mini E3 v1.2. Disabled the sanity check and pointed at @ellensp's neopixel fork, and it seems to work great.

DocYester, 2021-04-24:
Confirmed working on a MKS Robin E3D V1.0 (003)

bingston, 2021-05-10:
Having trouble getting Neopixels to work with me SKR Mini E3 v2.  I just have a small strip of 8-LED's to put near the nozzle, and I can't quite figure it out.  Has anyone else gotten it to work?  I'm fairly new to this so I'm not sure if/what I'm doing wrong.  Looking back at my compilation of Marlin f 2.0.7 the platformio.ini looks different and had Neopixels listed and you could change the path to the library.  I got the bugfix of Marlin 2.0.8 and I don't see that and I'm not sure how to add/change libraries.  Could someone help me please?

bingston, 2021-05-10:
ok, thanks!

Floppy, 2021-05-16:
@bingston they're not supported, but with a couple of code tweaks, they can be made to work. Read back through the comment thread here, a number of us have it working. The best solution right now is to change the neopixel library to the fork by @ellensp.

ellensp, 2021-05-16:
@floppy only unreliably they take to many resources away from Marlin, the more LEDs you have the worse it gets.

thisiskeithb, 2021-07-23:
Boards under STM32F1/maple are being migrated to the STM32 HAL, so this feature request will no longer apply as NeoPixels work under the STM32 HAL.

closed 2021-07-23 by thisiskeithb

Floppy, 2021-07-23:
@thisiskeithb that's fantastic news, thanks!

00ihsan, 2021-08-07:
Anybody knows when this is released? I am using a non maple version 2.0.9.1 for my btt skr mini E3 v2 but it is super laggy and the performance of my printer has dropped.