MarlinFirmware/Marlin #19802: [BUG] TMC Connection Error using original TMC library

Loreagle, 2020-10-18:
### Bug Description

If I enable hardware UART on my MKS Robin Nano V1.2 I have TMC Connection error but when I switch to https://github.com/bigtreetech/TMCStepper/ it works perfectly without errors.

### My Configurations

https://github.com/Loreagle/Marlin/tree/Sapphire-Pro

### Steps to Reproduce

1. Compile Marlin with original TMC libraries and hardware UART enabled
2. Flash firmware 

**Expected behavior:** 

Drivers works correctly

**Actual behavior:** 

TMC connection error is displayed

#### Additional Information

When I switch to https://github.com/bigtreetech/TMCStepper/ it works perfectly without errors and also M122 shows everything okay.

sjasonsmith, 2020-10-18:
Can you enable TMC_DEBUG and post your M122 output using the default library?

thisiskeithb, 2020-10-19:
> https://github.com/Loreagle/Marlin/tree/Sapphire-Pro

Do you have a different link for your config? The linked config appears to be a default with nothing specific to the Robin Nano 1.2 motherboard or TMC drivers.

sjasonsmith, 2020-10-19:
@Loreagle it would be best to zip up your configs and attach them. You should include the two config headers and anything else you've modified, such as platformio.ini and your pins file.

ellensp, 2020-10-19:
Full logs of compile might also help, would identify all versions of frameworks and libraries

Loreagle, 2020-10-19:
> Can you enable TMC_DEBUG and post your M122 output using the default library?

Sure, here you can see 3 different M122 outputs, as you can see I have bad response and connections errors randomly:
[image]
[image]
[image]

> > https://github.com/Loreagle/Marlin/tree/Sapphire-Pro
> 
> Do you have a different link for your config? The linked config appears to be a default with nothing specific to the Robin Nano 1.2 motherboard or TMC drivers.

I'm sorry I forgot to commit, now is updated!

> @Loreagle it would be best to zip up your configs and attach them. You should include the two config headers and anything else you've modified, such as platformio.ini and your pins file.

Sure! [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5400147/Configs.zip)

> Full logs of compile might also help, would identify all versions of frameworks and libraries

[Compile log.txt](https://github.com/MarlinFirmware/Marlin/files/5400164/Compile.log.txt)

As I said simply switching library with TMCStepper=https://github.com/bigtreetech/TMCStepper/ fix the UART connection problem and I have no errors, the other problem I have is an annoying coil whine when motors are in StealtChop mode only

Loreagle, 2020-10-20:
SOLVED!

I solved adding this line:

  // Reduce baud rate to improve software serial reliability
    #define TMC_BAUD_RATE 19200

It was only for software UART but apparently it fixes also hardware... I don't know why with BigTreeTech library it was working, maybe this line was inside the library?

sjasonsmith, 2020-10-20:
Something is really wrong if you need that for HardwareSerial. I’ve run HardwareSerial with TMC drivers at 500 kbaud.

sjasonsmith, 2020-10-20:
I just looked at your pins file and you are absolutely not using HardwareSerial. If you are defining pins and not port names you are using SoftwareSerial.

Loreagle, 2020-10-20:
> I just looked at your pins file and you are absolutely not using HardwareSerial. If you are defining pins and not port names you are using SoftwareSerial.

What, so how can I use Hardware serial? I still need to define where my cables are connected since my board doesn't support UART natively. I'm using hardware since addresses are settled via jumpers no?

closed 2020-10-20 by Loreagle