MarlinFirmware/Marlin #19804: [BUG] UBL with PROBE_MANUALLY crashes into the bed when "Mesh Edit" or "Fine Tune All" or "Fine Tune Closest" from LCD

qurvento, 2020-10-18:
### Bug Description

UBL with PROBE_MANUALLY crashes into the bed when running "Mesh Edit" or "Fine Tune All" or "Fine Tune Closest" from LCD. The printer tries twice to set Z=0 before going to Z of the mesh point, and crashes into the bed if current point's Z>0.

### My Configurations

Creality Ender-3 v4.2.2
Marlin 2.0.7.1
AUTO_BED_LEVELING_UBL, PROBE_MANUALLY, LCD_BED_LEVELING, MESH_EDIT_MENU, UBL_MESH_EDIT_MOVES_Z
[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/5398203/cfg.zip)

### Steps to Reproduce

1. Select "Motion/Unified Bed Leveling/Mesh Edit" or "Motion/Unified Bed Leveling/UBL Tools/Edit Mesh/Fine Tune All" or "Motion/Unified Bed Leveling/UBL Tools/Edit Mesh/Fine Tune Closest".

**Expected behavior:** moves to mesh point's Z

**Actual behavior:** moves twice to Z=0 before moving to mesh point's Z.

boelle, 2020-10-20:
is this related to #19814 ?

qurvento, 2020-10-24:
no

sjasonsmith, 2020-11-21:
Don't be fooled by the "Merged" PR above, that was a similar issue that I thought might have shared a root cause with this. It is very similar, but in different code.

sjasonsmith, 2020-11-22:
@qurvento I've posted a PR with your recommended changes. Can you look at it and verify that it works how you would expect?

Foxies-CSTL, 2020-11-22:
Hi, 
I noticed the same behavior on a Delta when I edit a mesh using the menu.
The two XY axes are positioned while Z reaches a point near 0 at high speed (I cut before the power supply).
Just before the power cut, I see the mesh screen with a square of dots.
[image]

sjasonsmith, 2020-11-23:
@Foxies-CSTL I don't quite understand what you are desribing. Is it directly related to this issue or the PR that was just merged? This issue involves the nozzle pressing into the bed. It sounds like your issue is your carriage moving to the side? I don't think my recent change could have caused this.

Foxies-CSTL, 2020-11-23:
>  I don't quite understand what you are desribing

Okay I thought the problem is pretty well described so I'm going to do a PR to fix the bad behavior of the Z axis stepper motor when I want to edit a mesh through the menu in UBL Tools.

> I don't think my recent change could have caused this.

I thought your PR would solve my problem but it's still there.

sjasonsmith, 2020-11-23:
@Foxies-CSTL wow, I just noticed that one carriage at the bottom. I missed that in the original picture. That is definitely not related to this issue.

Foxies-CSTL, 2020-11-23:
> That is definitely not related to this issue.

No problemo ;-) I write a PR for that =>#20257

sjasonsmith, 2020-11-28:
Closing this as resolved, since we haven't heard back from @qurvento after the merged PR.

closed 2020-11-28 by sjasonsmith