MarlinFirmware/Marlin #19807: [BUG] Setting SKR Pro 1.1 fan misbehavior

Borillion, 2020-10-18:
### Bug Description

With #define EXTRUDER_AUTO_FAN_TEMPERATURE 50 configured 
the fan doesn't turn on at 55 deg C 

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5398330/Configuration.zip)

### Steps to Reproduce

#define FAN_SOFT_PWM 
#define SOFT_PWM_SCALE 2

**Expected behavior:** [What you expect to happen]

With #define EXTRUDER_AUTO_FAN_TEMPERATURE 50 configured along with #define EXTRUDER_AUTO_FAN_SPEED 255  the hotend fan should turn on at 55 deg C 

**Actual behavior:** [What actually happens]

Fan remains off its connected to FAN1_PIN and should turn on.

thisiskeithb, 2020-10-18:
This is not a Marlin bug. You have `E0_AUTO_FAN_PIN` set to `-1` which disables it.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-18 by thisiskeithb

jirafb, 2020-10-28:
hi all

The AUTO_FAN feature **doesn't work** on Bigtreetech SKR PRO v1.1 even if I remove the **#define E0_AUTO_FAN_PIN -1** line.
The related pins definition file ("pins_BTT_SKR_PRO_common.h") states in such a case:

#ifndef E0_AUTO_FAN_PIN
  #define E0_AUTO_FAN_PIN               FAN1_PIN
#endif

And this causes just further malfunction, as decribed below.
BTT SKR PRO has three fans connectors, marked as FAN0 (physically connected to PC8), FAN1(PE5) and FAN2(PE6). 
This is matching to definition within pins_BTT_SKR_PRO_common.h file:
#define FAN_PIN                             PC8   // Fan0
#define FAN1_PIN                            PE5   // Fan1
#define FAN2_PIN                            PE6

so far so good, although I don't understand the mismatch in naming here - why "FAN_PIN" instead of "FAN0_PIN"??
If I leave the #define E0_AUTO_FAN_PIN -1 line in place, I can access under TEMPERATURE menu all three fans and control them manually. But the extruder fan does not start automatically over 50°C. 
When I comment out the line //#define E0_AUTO_FAN_PIN -1 , I can observe following consequence:
1.  the #ifndef condition assigns autostart functionality to Fan2
2.  above 50°C my Fan2 starts. But Fan2 is not the extruder fan! 
3. on top of that the extruder fan does disapear from TEMPERATURE menu, hence I can't control it even manually.

When I tried this definition (which shall be the correct one, I guess): 
#ifndef E0_AUTO_FAN_PIN
  #define E0_AUTO_FAN_PIN               FAN_PIN
#endif

... I got couple of error messages during compilation, that I can't assign FAN_PIN to E0_AUTO_FAN_PIN.

**How can I bring my extruder fan (physically driven by PC8) to automatical run over 50°C ??**

Thanx in advance & best regards
Frodo

thisiskeithb, 2020-10-28:
@jirafb Use the resources in my comment above for config troubleshooting.