MarlinFirmware/Marlin #19808: [FR] #define for extruder cooling fans to select board's pin file as defaults

Borillion, 2020-10-18:
### Description

It's somewhat ambiguous from the comments in `@section extruder : Extruder Cooling fans` what the expected behavior for someone who is doing a build should do; and what spurred me to create a bug request out of my not grasping the concepts. https://github.com/MarlinFirmware/Marlin/issues/19807 

Would it difficult to add a define or other if else function https://github.com/MarlinFirmware/Marlin/blob/0a0f211cb9faa9213abff3ec45b3c21b4b0d499a/Marlin/Configuration_adv.h#L466 that enables what is set in the boards pins file?

thisiskeithb, 2020-10-18:
I'm not sure I follow. This is already in place and the readme above the section explains what to do:

https://github.com/MarlinFirmware/Marlin/blob/6375829448eebe01b2835689aac69b73b63ebfd8/Marlin/Configuration_adv.h#L471-L472

They are disabled by default because not all boards support automatic fans/the default behavior is for the extruder fan to be connected to an always-on power source (so no auto fans).

borland1, 2020-10-20:
I have `E0_AUTO_FAN_PIN` pin assigned (enabled) on my printer and it seem to run as expected, however when I press the encoder knob to make a LCD menu selection, the fan power is cut momentarily.  Is this expected behavior? I don't recall seeing any comments about that.

My board is based on AVR Mega 2560 and would expect the fan is running on `glitch-free` timer/counter PWM, so Auto_FAN should run on this machine independent of LCD menu selections.  Apparently, the AUTO_FAN PWM timing  is not controlled entirely by the associated counter/timer.

thisiskeithb, 2020-10-20:
@borland1: Please file a bug report since that is unrelated to this feature request.

borland1, 2020-10-20:
OK, but I thought it was related to this feature request.  I can't file a bug report because even after reading all the docs and source code comments, I'm not sure what I'm seeing is actually a bug or expected behavior?

closed 2022-05-10 by thisiskeithb