MarlinFirmware/Marlin #19810: Freezing print randomly New ender3 with the 4.2.2 board

kiritokunsama, 2020-10-19:
# Ender3 with the 4.2.2 board randomly stops during prints

All was working fine but suddenly started stopping during the prints. 
Everything was on (motors/fans) and hot (hot end/bed). The LCD screen was normal like I was printing.

I pressed the main button and suddenly started printing normally. I'm using the last marlin bugfix (#19765 0a0f211)

My Marlin2 files [link](https://gofile.io/d/2U2R8u)

It is my first issue that I post here in Github so take it easy and sorry if I did something wrong

BuzzDaClown, 2020-10-19:
Following

kiritokunsama, 2020-10-19:
I noticed a "click" sound that happens sometimes (the same sound when u click the button). And, apparently, it is auto-clicking because I left it on the "info screen" and after the click, when I came nearby, It was on the "options screen".

Nu-hin, 2020-10-19:
The "click" sound problem also happened to me, and, according to Reddit to numerous users of Ender 3, Ender 3 Pro and Ender 5 with stock mainboards and with custom one. 

As suggested [here](https://www.reddit.com/r/ender3/comments/ejgv3x/ender_3_ender_5_pushing_button_by_itself_while/), this can be connected to an electric leakage.

In my case I could radically solve it by detaching my complete LCD module from the printer base, so that metal parts (LCD case and printer base) don't touch.

Nu-hin, 2020-10-19:
Exactly the same problem with pausing prints started happening to me since yesterday, when I upgraded my Ender 3 Pro (4.2.2 board) from stock firmware to #19359 (bugfix-2.0-X).

It pauses on longer prints, randomly, sometimes 5 minutes between pauses.

Hotend and bed stay heated, and the nozzle stays in the plastic, making an ugly blob.

When I press the knob, the print instantly resumes.

I also noticed that a screen during these pauses shows a couple of weird characters, where printed filename used to be. After pushing the knob, the filename is displayed again.

I am printing from SD card, and I have checked the Gcode for anything suspicious - could not find anything.

kiritokunsama, 2020-10-20:
Exactly Nu-hin! I think there is another problem, but I'm still working on it.

Nu-hin, 2020-10-24:
@kiritokunsama Do you print from SD card? Have you tried replacing it? I had exactly the same symptoms as you described, then I replaced the poor SD card which came with the printer with a more expensive Samsung one. Since then I have completed five 7-8 hour print and never run into this issue again.

sjasonsmith, 2020-10-25:
I'm labeling this with Safety and LCD labels, since it seems it is waiting for a user-interaction through the LCD button?

I thought I was reading about a completely hung printer, but that seems not to be true if it resumes eventually.

kiritokunsama, 2020-10-25:
> @kiritokunsama Do you print from SD card? Have you tried replacing it? I had exactly the same symptoms as you described, then I replaced the poor SD card which came with the printer with a more expensive Samsung one. Since then I have completed five 7-8 hour prints and never run into this issue again.

I've tried other SD cards but still the same here.

kiritokunsama, 2020-10-25:
> I'm labeling this with Safety and LCD labels, since it seems it is waiting for a user-interaction through the LCD button?
> 
> I thought I was reading about a completely hung printer, but that seems not to be true if it resumes eventually.

Actually, it freezes completely the printer (the motion).

Nu-hin, 2020-10-25:
@kiritokunsama I think what's important is that it reacts to the main button, as you wrote:

> I pressed the main button and suddenly started printing normally.

For me it was the same.

kiritokunsama, 2020-10-26:
> @kiritokunsama I think what's important is that it reacts to the main button, as you wrote:
> 
> > I pressed the main button and suddenly started printing normally.
> 
> For me it was the same.

I agree, and the "auto-click" acts like someone pressed the main button.

kiritokunsama, 2020-10-26:
Updated the post.

kiritokunsama, 2020-11-26:
Freezing SOLVED. Just Updated again to the newer version

rhapsodyv, 2020-12-02:
I'm closing this, as the user said the it's fixed when updating new newer version.

closed 2020-12-02 by rhapsodyv