MarlinFirmware/Marlin #19813: [FR] Option to set M85 S0 if print is manually stopped from printer's menu

bkiziuk, 2020-10-19:
### Description

I'm using `M85 S60` command in start-G-code sequence to protect against potential print freezes.
In end G-code I reset the timer with `M85 S0`.

However, during ongoing print, when the printing is manually stopped by choosing "Stop print" option (operator's intervention), the kill timer is not reverted to 0 so that the printer halts itself after the timer runs out. At that stage the printer must be rebooted.

### Feature Workflow

1. Set configuration option (something like M85_STOP_TIMER_ON_MANUAL_INTERVENTION) to stop kill timer on manual print interruption
2. Start print using `M85 Sxx` in the start G-code sequence
3. if the print is manually stopped (or paused?) from printer's panel, set `M85 S0` to avoid killing the software operation

#### Additional Information

None.

thisiskeithb, 2020-10-20:
You could use `EVENT_GCODE_SD_ABORT` for this:

`#define EVENT_GCODE_SD_ABORT M85 S0`

bkiziuk, 2020-10-21:
Thank you for an answer. I must have missed it. Is this also possible to execute a series of commands like:
```#define EVENT_GCODE_SD_ABORT "G0 X0 Y0 Z 100" "M85 S0"``` ?

ellensp, 2020-10-21:
use #define EVENT_GCODE_SD_ABORT "G0 X0 Y0 Z100\nM85 S0"

bkiziuk, 2020-10-21:
Many thanks for you help.

closed 2020-10-21 by bkiziuk