MarlinFirmware/Marlin #19814: [BUG] UBL with PROBE_MANUALLY ignores MESH_EDIT_Z_STEP

qurvento, 2020-10-19:
### Bug Description

UBL with PROBE_MANUALLY ignores MESH_EDIT_Z_STEP value and changes Z by 0.010, 0.015 or sometimes 0.020 mm per turn while creating the mesh via LCD.

### My Configurations

Creality Ender-3 v4.2.2
Marlin 2.0.7.1
AUTO_BED_LEVELING_UBL, PROBE_MANUALLY, LCD_BED_LEVELING
[cfg.zip](https://github.com/MarlinFirmware/Marlin/files/5401329/cfg.zip)

### Steps to Reproduce

1. Select "Motion/Unified Bed Leveling/UBL Tools/Manually Build Mesh".

**Expected behavior:** step size is equal to MESH_EDIT_Z_STEP while manually probing Z axis

**Actual behavior:** step size varies and can be 10, 15 or 20 per turn of the rotary encoder

shitcreek, 2020-10-20:
In `menu_bed_leveling.cpp`, we find `MESH_EDIT_Z_STEP` nested inside of `#if EITHER(PROBE_MANUALLY, MESH_BED_LEVELING)`
We also see that further down we have
```
void menu_edit_mesh() {
    static uint8_t xind, yind; // =0
    START_MENU();
    BACK_ITEM(MSG_BED_LEVELING);
    EDIT_ITEM(uint8, MSG_MESH_X, &xind, 0, GRID_MAX_POINTS_X - 1);
    EDIT_ITEM(uint8, MSG_MESH_Y, &yind, 0, GRID_MAX_POINTS_Y - 1);
    EDIT_ITEM_FAST(float43, MSG_MESH_EDIT_Z, &Z_VALUES(xind, yind), -(LCD_PROBE_Z_RANGE) * 0.5, (LCD_PROBE_Z_RANGE) * 0.5, refresh_planner);
    END_MENU();
  }
```
So it looks like `MESH_EDIT_Z_STEP` is only used for `PROBE_MANUALLY` or `MESH_BED_LEVELING` and it is not used at all in the LCD mesh edit menu.

With `EDIT_ITEM_FAST` the faster you turn the knob, the greater the increment.

The other issue is that these encoders/dials/knobs we use are often not very reliable. Sometimes a single tick increments four times. If I am gentle enough I can turn the dial with or without incrementing.

qurvento, 2020-10-24:
I'm making only one fixed turn, and the result is always different. It never happens while moving manually any axis.

closed 2020-12-26 by github-actions[bot]