MarlinFirmware/Marlin #19816: [FR] Is it possible to disable S-Curve Acceleration temporarily during runtime?

yysh12, 2020-10-19:
### Description

Ability to toggle S-Curve Acceleration on and off during runtime without recompiling.

My use case: I am trying to calibrate the Linear Advance factor for each filament and the S-curve is affecting my ability to assess the calibration pattern generator on marlin website. It'd be nice to turn it off so I can get faster acceleration and more easily discern the correct K factor.

### Feature Workflow

Possibly edit M205 to allowing toggling of S-Curve. 

i.e. M205 C1 for on and C0 for off?

#### Additional Information

Use Case:

qwewer0, 2020-10-19:
Why don't just turn S-Curve acc. off, if you want to use Linear Advance? LA and S-Curve isn't 100% compatible with each other, S-Curve will throw off LA's pressure compensation.

yysh12, 2020-10-19:
Yeah you're probably right. I will toggle one or the other instead of trying to combine them. Thanks.

closed 2020-10-19 by yysh12

thisiskeithb, 2020-10-19:
Disabling S-curve to tune your LA value won't do you any good during an actual print.

yysh12, 2020-10-19:
Yeah. I just disabled S-Curve and went ahead with just LA. Print looks pretty good so imma just stick with it. Thanks.