MarlinFirmware/Marlin #19822: platformIO IDE fails to work

sebby49, 2020-10-20:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sebby49, 2020-10-20:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5407849/Marlin.zip)

ellensp, 2020-10-20:
How about some clues, how to reproduce this issue? error messages, logs, procedures.  You know all the tings the template asked for. Also I cannot extract your Zip file. All I get is zero length files.

thisiskeithb, 2020-10-20:
Your configs build fine with the latest `bugfix-2.0.x`:

```cpp
Linking .pio/build/STM32F103RET6_creality/firmware-20201020-024922.elf
Checking size .pio/build/STM32F103RET6_creality/firmware-20201020-024922.elf
Building .pio/build/STM32F103RET6_creality/firmware-20201020-024922.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.0% (used 10488 bytes from 65536 bytes)
Flash: [====      ]  37.9% (used 198828 bytes from 524288 bytes)
======================================== [SUCCESS] Took 20.73 seconds ========================================

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  SUCCESS   00:00:20.733
======================================== 1 succeeded in 00:00:20.733 ========================================
```

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-20 by thisiskeithb