MarlinFirmware/Marlin #19823: Dual Z not working on skr pro v1.1

oscar6502, 2020-10-20:
### Bug Description

After bringing the last changes in the bugfix branch, the z2 driver does not work.
Controller skr pro v1.1
driver tcm2130 spi

Sorry for my English, I use a translator.

### My Configurations

Configuration.h

#define X_DRIVER_TYPE  TMC2130
#define Y_DRIVER_TYPE  TMC2130
#define Z_DRIVER_TYPE  TMC2130
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2130
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2130
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

Configuration_adv.h

#define NUM_Z_STEPPER_DRIVERS 2 

#define Z_STEPPER_AUTO_ALIGN

  #ifndef Z_STEPPER_ALIGN_XY
    #define Z_STEPPERS_ORIENTATION 0
  #endif

### Steps to Reproduce

1. git pull
2. compile
3. upload

**Expected behavior:** working
**Actual behavior:** not working

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

oscar6502, 2020-10-20:
> Bad timing. A bug was introduced into bugfix yesterday, the fix is here #19820 but not yet part of marlin bugfix

Excellent, thank you for your comment!

x-c-dog-x, 2020-10-20:
Not working for SKR v1.4 also. Found out the hard way, because bugfix requires a new CONFIGURATION_ADV_H_VERSION 020008

cropduster, 2020-10-20:
IOn my dual z configuration the both axis do not travel the same way. Is that this particular issue? It just say sit is not working. I have a SKR 1.4 Pro board with TMC2209 drivers.

thisiskeithb, 2020-10-20:
@x-c-dog-x & @cropduster: Apply the patch from https://github.com/MarlinFirmware/Marlin/pull/19820 to the latest `bugfix-2.0.x`. This has been confirmed to work for everyone who reported the issue originally.

thisiskeithb, 2020-10-20:
Closing since PR #19820 was merged.

closed 2020-10-20 by thisiskeithb