MarlinFirmware/Marlin #19826: [BUG] Marlin loses control of hotend during long print

rq3, 2020-10-20:
During a 7 hour print under 2.0.7.2, the printer consistently loses PID control of the hotend temperature at hour 5. This was repeatable over 3 prints. At the fifth hour (300 minutes, irrespective of layer or other factors), the hotend temperature enters a slow yet increasing oscillation mode until the printer freezes with an E1 temperature error (hotend temperature in excess of 10C above or below set point).

Please don't tell me to do a PID check, or verify that the sensor is OK. Been there. Done that. Modifying configuration_adv.h to allow this is also unacceptable, as a 20 degree swing in nozzle temperature is not OK during any print.

ellensp, 2020-10-21:
Please attach your configuration files.

SJ-Innovation, 2020-10-21:
Does this occur when not printing? As in, (youll want to take out the filament so it doesnt burn in) leave the hotend on for 5 hours without printing.

boelle, 2020-10-22:
will temp close this one until configs are provided

thisiskeithb, 2020-10-23:
@rq3 Please continue troubleshooting here instead of creating multiple issues for the same thing.

Also: Please do as everyone asked multiple times and attach your config files.

thinkyhead, 2020-10-23:
I don't think this is a bug in Marlin. Something that occurs like this, after 5 hours of printing, sounds very much like some kind of failing electronic component. PID loops are clever, but they aren't complicated, and they don't just fall apart after a 5 hour period of stability. Seek help in the various forums until it is clear from thorough isolation testing that a bug in Marlin is the only possible cause.

closed 2020-10-23 by thinkyhead