MarlinFirmware/Marlin #19829: [FR] (Choose BLTOUCH/Probe for Y-Axis)

Rizzist, 2020-10-21:
### Description

Basically, I have a belt printer and the Y-Axis is the conventional Z axis in a regular printer. I usedd BLTOUCH on the Y axis for the printer and it didnt work, mainly because of the fact that BLTOUCH is only for Z-AXIS. Support for this would be nice.
### Feature Workflow

1. Turn on Y Axis Probe
2. Use like regular sensor

qwewer0, 2020-10-21:
Could you describe it a bit more, to what do you mean Y axis is the Z axis in your setup?

Rizzist, 2020-10-21:
> Could you describe it a bit more, to what do you mean Y axis is the Z axis in your setup?
> Normally the Z axis is for vertical moves, and X/Y axis are for horizontal moves.

Yes, in this case its flipped, where Y axis for vertical moves, X/Z for horizontal moves. Im not sure what else to add to the description other than having another axis have BLTOUCH possible.

qwewer0, 2020-10-21:
Can't you just switch the Y and Z stepper cables?

Do you have a picture of that kind of printer or a link to a product page?

Rizzist, 2020-10-21:
> Can't you just switch the Y and Z stepper cables?
> 
> Do you have a picture of that kind of printer or a link to a product page?

The issue with that is because the slicer does not have a way to switch Y and Z axis, its using Blackbelt Cura. I tried vigorously to find a way like writing a python script to replace Y and Z gcode but its not possible unless i use a regular switch instead of probe.

Rizzist, 2020-10-22:
its a homemade belt printer, kind of like blackbelt 3d printer
[image]

Rizzist, 2020-10-22:
I would really like to have this a feature, I got the belt printer working with the regular limit switch, but I have to remove it after leveling and I am constantly afraid that it will crash, and highly prefer the BLTOUCH.