MarlinFirmware/Marlin #19832: [BUG] example configuration file for FLSUN Q5

olinek2, 2020-10-21:
Hi, In Configurations-release-2.0.7.2\config\examples\delta\FLSUN\Q5-nano_v1\Configuration.h file there is a bug in Lines 499, 500, 501.
Actual version:
```
 #else
    // flsun Q5 via M303 C8 E-1 S60
    #define DEFAULT_bedKp 80.77
    #define DEFAULT_bedKi 15.74
    #define DEFAULT_bedKd 276.27
  #endif
```

It should be:
```
  #else
    // flsun Q5 via M303 C8 E-1 S60
    #define DEFAULT_Kp 80.77
    #define DEFAULT_Ki 15.74
    #define DEFAULT_Kd 276.27
  #endif
```
Best regards,
Olek

alexeyzel, 2020-10-21:
I think values should be like this:
`    #define DEFAULT_Kp 12.88
    #define DEFAULT_Ki 0.72
    #define DEFAULT_Kd 57.54`

olinek2, 2020-10-21:
I've tested 
    #define DEFAULT_Kp 10.39
    #define DEFAULT_Ki 0.55
    #define DEFAULT_Kd 48.7

but it was to slow. Nevertheless the problem was not i the value, but in wrong defines, which led to bug in compilation.

ellensp, 2020-10-21:
This was fixed 15 days ago... https://github.com/MarlinFirmware/Configurations/issues/251

ellensp, 2020-10-21:
ok was meant to have been fixed...  something is wrong here...  not is import branch,,

ellensp, 2020-10-22:
Now its been fixed.

closed 2020-10-22 by ellensp