MarlinFirmware/Marlin #19833: [BUG] "Manual Probe" in "Dual X Carriage" Mode only XYZ Homing , cant 9 points probe.

xsuper9988, 2020-10-21:
[BUG] "Manual Probe" in "Dual X Carriage" Mode only XYZ Homing , cant 9 points probe.

boelle, 2020-10-21:
is #19835 the same problem as this one?

xsuper9988, 2020-10-22:
> #19835

Maybe some motion move bug at homing finish

and the bug likes #19835 

But when i did not use "Dual X Carriage" and "Z Safe Homing"

"Manual Probe" function is OK

xsuper9988, 2020-10-22:
Maybe some motion move function bugs at homing finish

and the bug likes #19835 

But when i did not use "Dual X Carriage" and "Z Safe Homing"

"Manual Probe" function is OK

sjasonsmith, 2020-11-18:
How does bed leveling work with an IDEX printer? You have two carriages on a single Z axis. It will be impossible to compensate for both at the same time.

@InsanityAutomation what are your thoughts on this?

closed 2020-12-26 by github-actions[bot]