MarlinFirmware/Marlin #19834: [BUG] Robin Nano v1.2 with Marlin Color UI display mirrored (left/right) on 3.2 screen

alexeyzel, 2020-10-21:
### Bug Description

After installing the latest Marlin release 2.0.7.2 on my Flsun Q5 with MKS Robin Nano v1.2 board with 3.2-inch screen I get my screen flipped (mirrored) horizontally.

### My Configurations

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5414866/Config.zip)
### Steps to Reproduce

1. Install Marlin via SD card
2. Try to turn printer off and on again

**Expected behavior:** 
Normal screen position.

**Actual behavior:** 
Screen is flipped horizontally.
#### Additional Information
May be related to https://github.com/MarlinFirmware/Marlin/issues/19603 but in my case printer rebooting don't help.
Looks like this:
[image]

ellensp, 2020-10-21:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

alexeyzel, 2020-10-21:
Thank you, I will test it today and write back.

thisiskeithb, 2020-10-21:
The `bugfix-2.0.x` branch has a `MKS_ROBIN_TFT32` define now and you shouldn't need to modify your pins file.

alexeyzel, 2020-10-21:
Tested in bugfix-2.0.x - screen is aligned properly. Thank you!

boelle, 2020-10-21:
@alexeyzel if problem solved please click green close button

closed 2020-10-21 by thisiskeithb