MarlinFirmware/Marlin #19835: [BUG] About "Z Safe Homing"...

xsuper9988, 2020-10-21:
My Ver 2.0.7.1

When i use "Dual X Carriage"
The situation is
1. XY homing
2. XY move to XY center
3. Z homing
"Z Safe Homing" normal work.

When i did not use "Dual X Carriage"
The situation is
1. XY homing
2. XY did not move to XY center
3. Z homing
4. Main page XY position is center,
    but machine XY axis is in Zero position

ellensp, 2020-10-21:
you have //#define Z_SAFE_HOMING

abacabie, 2020-10-21:
Just looked through your Configuration.h file and you don't have #define Z_SAFE_HOMING uncommented, just uncommented, compile and your good to go.

xsuper9988, 2020-10-21:
i test three times
when homing finish
machine xy axis in zero position
but main page xy in center position
and i use manual probe(#19833)
machine xy axis will be dangerous hit
so i mark //#define Z_SAFE_HOMING

boelle, 2020-10-21:
is #19833 the same problem as this one?

xsuper9988, 2020-10-22:
Maybe some motion move bug at homing finish

and the bug likes #19833 

But when i did not use "Dual X Carriage" and "Z Safe Homing"

"Manual Probe" function is OK

xsuper9988, 2020-10-22:
Maybe some motion move function bugs at homing finish

and the bug likes #19833 

But when i did not use "Dual X Carriage" and "Z Safe Homing"

"Manual Probe" function is OK

sjasonsmith, 2020-11-07:
I don't really understand what the issue is in this report. Which exact behavior do you consider incorrect?
Please post the _exact_ configuration files that you use to reproduce that behavior.

boelle, 2020-11-07:
@xsuper9988 ^^^^^

sjasonsmith, 2020-11-16:
It is still not clear to me what this issue is reporting. Since no additional detail was added after my last reply, I am going to close this as not actionable.

When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact. This will help make it much easier for others to understand your issue and provide help.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2020-11-16 by sjasonsmith