MarlinFirmware/Marlin #19838: [FR] Avoid __verbose_terminate_handler()

RealDeuce, 2020-10-21:
### Description

When building with PlatformIO for MKS Robin E3D, libstdc++ is compiled to use __verbose_terminate_handler, which pulls in all the symbol demangling code and increases the .text segment size by over 34KiB.  It would be nice if this was avoided.

### Feature Workflow

Saves 34KiB

1. Build without __verbose_terminate_handler
2. Size is smaller

#### Additional Information

As a proof of concept, I added the following to the linker script:
```_ZN9__gnu_cxx27__verbose_terminate_handlerEv = abort;```
And the used flash size went from 237444 to 202340

Obviously this is a terrible and wrong way to do it, but you get the idea.

RealDeuce, 2020-10-21:
Note, normally, you would configure libstdc++ with --disable-libstdcxx-verbose but since the library comes from a PlatformIO package, that doesn't seem like a viable option.

sjasonsmith, 2020-10-25:
Perhaps since you are already looking into this you should submit a pull request with the change?
I'm not sure if there are any negative consequences of the change, but putting it up in a PR could spur discussion on the topic.

RealDeuce, 2020-10-25:
I guess I could, the main reason I didn't is that I usually think of a pull request as "this is should be merged" rather than opening a discussion.  What I did as a proof of concept is basically the nuclear option and (IMO) absolutely should not be merged... it relies on the name mangling and subverts the standard C++ library.

Maybe using a stub wrapper around __verbose_terminate_handler() would work, I'd need to test it out... but that would still be a poor choice.  The proper solution is to build libstdc++ with different options, which appears to mean working with the PlatformIO infrastructure, something I'm completely unfamiliar with.  It's possible that there's a simple way to get the needed libstdc++ from PlatformIO.

qwewer0, 2020-10-25:
@RealDeuce You could open the PR as draft.

sjasonsmith, 2020-10-25:
@RealDeuce would you mind looking at this for the mks_robin_stm32 environment to see if it has the same size issue? It likely won’t immediately work with your board, but it uses the ST framework instead of Maple.

Someone recently mentioned that it already uses a different version of libc, which is much smaller.

If that framework doesn’t have the same issue, then perhaps time is better spent accelerating the migration from Maple to ST for all boards.

RealDeuce, 2020-10-27:
Confirmed that mks_robin_stm32 does not use the verbose termination handler.

sjasonsmith, 2020-12-01:
@RealDeuce we are seeing a similar size reduction from updating to a newer version of Maple, as is currently pending in PR #20325. You may want to look at that PR to see if the new platform is already doing this. I am not sure if this is where the size improvement came from, or if it is from some other change.

sjasonsmith, 2020-12-02:
We just updated to use a new Maple version which reduces FLASH usage by ~30kB, similar to this recommendation.

At this point we don't want to invest additional effort into more Maple improvements. We need to focus on migrating our STM32F1 boards into HAL/STM32, so that the Maple dependency can be eliminated entirely.

For this reason I'm going to close this out, even if there might be some potential for further improvements.

closed 2020-12-02 by sjasonsmith