MarlinFirmware/Marlin #19839: [BUG] BLTouch not compensating

gbkwiatt, 2020-10-21:
### Bug Description
I used to have Ender 3 with v1.1.5 board and BL touch on marlin 1.1.8 and never had to level the bed, was always perfect thanks to bltouch. 
Now, I have Ender with Bltouch again, but with V427 board with BLtouch plugged to Bl connector on the board. 
 No matter what I do, despite it looks all okay, BL touch is detecting bad level (bad visualisation)  it's not compensating for bed level. I know I could level it manually, but that's not a case. And my level differences are minimal so Bltouch should easily compensate it. I do feel like Z axis is moving when moving around bed, but probably not enough. 
I tried to find issues anywhere else, took printer apart , changed bed, changed filament, slicer everything ... but it's always same result. 
I don't know how else I can debug it to make sure the compensation is right ? 

### My Configurations
Ender 3 + Creality BLTouch 
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5418616/Marlin.zip)
Latest BugFix Marlin with latest  config files (compiled 10 minutes ago from repo)

**Expected behavior:** Compensate for bed mesh detected by bltouch

**Actual behavior:** No compensation for unleveld bed regardless bltouch

shitcreek, 2020-10-21:
Have you tried following this guide: https://www.youtube.com/watch?v=neS7lB7fCww ?

gbkwiatt, 2020-10-21:
> Have you tried following this guide: https://www.youtube.com/watch?v=neS7lB7fCww ?

That's a guide for V2 and I have normal ender 3, besides I checked all guides and setups on the Internet for v427, you cant do much wrong with it. Also like i mentioned BLtouch works and bed visualisation proves that
[image]

It's not big unevenness, BLtouch should easily compensate it, but it doesn't... When I set z offset for 1 corner nicely, it's not right for other corners.

shitcreek, 2020-10-22:
This probably isn't the issue, but are you using the Creality BLTouch or ANTCLABS BLTOUCH? Because you've got `BLTOUCH` enabled instead of `CREALITY_TOUCH`.

ellensp, 2020-10-22:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Talha909, 2020-10-22:
I have the same issue. I have skr 1.4 Turbo with tmc 2130 and ANTCLABS BLTOUCH. Its home perfectly, after that I gave the g29 command and its probe on 9 points. And when prints start there is no compensation.

ellensp, 2020-10-22:
@Talha909 without providing config files and logs saying "I have the same issue" is rather pointless.

gbkwiatt, 2020-10-22:
> This probably isn't the issue, but are you using the Creality BLTouch or ANTCLABS BLTOUCH? Because you've got `BLTOUCH` enabled instead of `CREALITY_TOUCH`.

I did see that and tried but with CREALITY_TOUCH my bltoch is not deploying at all. I have BLtouch from creality

gbkwiatt, 2020-10-22:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:
> 
Will do it and report, should I also do print with tha kgo enabled, as that's where I can only see a problem, during print

thisiskeithb, 2020-10-22:
`CREALITY_TOUCH` is for the hotend sensor used on the CR6-SE.

You’d still use `BLTOUCH`.

thisiskeithb, 2020-10-22:
Is this a duplicate of https://github.com/MarlinFirmware/Marlin/issues/19845?

gbkwiatt, 2020-10-22:
> Is this a duplicate of #19845?

No, that's not related to this issue. I am using BLTOUCH for that issue

gbkwiatt, 2020-10-22:
here is log: 
[octo_log.txt](https://github.com/MarlinFirmware/Marlin/files/5422384/octo_log.txt)

And I have notice something hmm weird ? 

so that;s the final output of G29:
```
Recv:       0      1      2
Recv:  0 -0.025 +0.140 +0.237
Recv:  1 -0.160 -0.020 +0.033
Recv:  2 -0.290 -0.180 -0.158
```

Is 0:0 the 0x0 position of the bed? so in ended 3 case bottom left corner ? 
if so 0:2 = -0.290 should be bottom right corner ?  
If yes, the this is wrong, because my bottom right corner according to bed visualiser is positive up, highest point which should be 0.237

gbkwiatt, 2020-10-22:
I have another hint, i leveled bed manually and first layer Perfect Bur second too low...
Which means there is an issue with Z axis, is there any, step calibration for steppers or something?

gbkwiatt, 2020-10-23:
So I did test, and put my old 1.1.4 board to ender 3 with bltouch and with marlin from creality. 
I super messed up with bed: 
[image]

And it printed fine ... didn't even touch the bed. 
So I must say problems with V427 are something with firmware well or the board itself ?

gbkwiatt, 2020-10-24:
I've testes everything throughly, and I can confirm that BLtouch works as intended on marlin 1.1.9, and hoard 1.1.4, but Unfrotunatley marlin 1.1.9 does not work on v427 board so can't test it, but seems like the issue is with marlin 2.
Is there any plans too look into that issue? I don't want to be stuck with old loud board

Liger0, 2020-10-24:
Is this issue maybe related to #15450 even if the hardware is different? 
Symptoms are the same, the bed is correctly checked but there is no compensation.

gbkwiatt, 2020-10-25:
It is same issue by the look of it.

Sebazzz, 2020-10-26:
> 
> 
> `CREALITY_TOUCH` is for the hotend sensor used on the CR-6 SE.
> 
> You’d still use `BLTOUCH`.

No, it is not the hot-end sensor of the CR-6. In fact, in the Creality firmware, CREALITY_TOUCH is not defined. If you define CREALITY_TOUCH, BLTOUCH is defined automatically for you.

sjasonsmith, 2020-10-26:
> No, it is not the hot-end sensor of the CR-6. In fact, in the Creality firmware, CREALITY_TOUCH is not defined. If you define CREALITY_TOUCH, BLTOUCH is defined automatically for you.

It uses the BLTOUCH code, but it doesn't behave exactly the same. If you have something that looks like a BLTouch on your system, do not define CREALITY_TOUCH. It will not work properly.

It doesn't matter what Creality does in their own firmware. Vendors are notorious for changing all sorts of things, and you can't always directly compare their configurations to mainline Marlin.

If you want proof that they are not synonyms: (from `bltouch.h`)
```
#if ENABLED(CREALITY_TOUCH)
  #define STOW_ALARM         false
  #define BLTOUCH_DEPLOY       170
  #define BLTOUCH_STOW          20
#else
  #define STOW_ALARM          true
  #define BLTOUCH_DEPLOY        10
  #define BLTOUCH_STOW          90
#endif
```

Defining `CREALITY_TOUCH` uses completely different PWM pulse widths when comapred to `BLTOUCH`.

gbkwiatt, 2020-10-26:
That's out of scope that issue guys

Talha909, 2020-10-31:
did you fine any solution

gbkwiatt, 2020-10-31:
No, Spending nights and days on it and I can't make it work

gbkwiatt, 2020-10-31:
I have literally tried everything, took my printer apart couple of times, to makes sure everything is super correct. and it still does not compensate for not leveled bed. I can level it manually spot on and that's fine, but I want BLtouch to do it for me.

CRCinAU, 2020-11-08:
Stupid question - but not mentioned here yet... Have you enabled the mesh via M420 S1 in your start gcode?

gbkwiatt, 2020-11-08:
No, but there is no need for that because G29 does it for you

sjasonsmith, 2020-11-29:
I tested with almost your exact configs. It seemed like there might be an issue, but that also may be because you do not have `EXTRAPOLATE_BEYOND_GRID` enabled. I tested with a bed about 6mm off from edge to edge. While correction wasn't perfect, it also was not off by 6mm.

The reformatting you did in the config files makes it very difficult to tell what you changed. I intend to try again tomorrow with only the minimal changes made to the example.

tarsonis123, 2020-11-29:
Same here. Creality 4.2.7 on Ender with 3 BLtouch Clone. Unreliable nightmare since switching from 1.1.5 and stable firmware that dumb enough forgot to save. Since the 4.2.7 and the bugfix branch I'm playing beta tester not printing. Ending up with manually editing all points where it makes the BL obsolete. BLtouch works with 0.003 deviation and was flawless with 1.1.5 and an stable Marlin 2.x release.

sjasonsmith, 2020-11-30:
I spent several hours testing this today. Due to the difficulty of spotting changes in the reformatted configuration files, I chose to start with the Ender 3 v427 example, and add the BLTouch configuration.

I am using a Creality 4.2.7 board with a genuine BLTouch attached to the dedicated probe connector. I have a BMG extruder, so I have adjusted the E0 steps and inverted its direction accordingly.

I am testing with the `bugfix-2.0.x` branch, specifically on top of commit df0a0c9490346f57e1b8224cb0fe7a1ad46d76bb.

All of my changes from the example are summarized below:
```
// #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define USE_PROBE_FOR_Z_HOMING
#define BLTOUCH
#define NOZZLE_TO_PROBE_OFFSET { -48, -10, -2.1 }
#define Z_PROBE_LOW_POINT          -10 // This is extreme, because I tested with extremely un-level bed
#define INVERT_E0_DIR true
#define AUTO_BED_LEVELING_BILINEAR
#define GRID_MAX_POINTS_X 5  // It worked with 3, but wasn't quite enough resolution to correct one point on my bed
#define EXTRAPOLATE_BEYOND_GRID
#define Z_SAFE_HOMING
```

Here are the configs themselves: 
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5613842/Configs.zip)

I made my bed very uneven, producing the following mesh, with >6mm deviation across the bed.
```
Bilinear Leveling Grid:
      0      1      2      3      4
 0 -3.322 -2.435 -1.517 -0.560 +0.500
 1 -2.770 -1.887 -0.975 -0.017 +1.135
 2 -2.277 -1.375 -0.427 +0.582 +1.710
 3 -1.660 -0.735 +0.193 +1.237 +2.332
 4 -1.020 -0.102 +0.895 +1.907 +2.965
```

I then printed a 1-layer test pattern that touches all regions of the bed. There are a couple minor adhesion issues and my nozzle was too close to the bed. You can see that each of the patches has the same texture, indicating that bed leveling behaved as expected. Remember that this is correcting over 6mm of deviation across a 235mm bed!

[image]

sjasonsmith, 2020-11-30:
At this point I don't believe there is any problem in Marlin or with the Creality 4.2.7 board causing this.

I recommend that you switch back to the example configurations, with minimum changes to enable leveling. You can use my configs from above, just remember to revert the E0 changes. _Please do not reformat the file, since this makes it extremely difficult to compare against default values._

In addition, please enable `Z_MIN_PROBE_REPEATABILITY_TEST` to verify whether you are getting consistent probe readings. People often mention standard deviation with these results, but I prefer to look at range.

These are my results with 20 probes:
```
M48 P20
...
Mean: 0.003375 Min: -0.007 Max: 0.033 Range: 0.040
Standard Deviation: 0.010934
```

That range is a little higher than I was expecting, although I expect even 0.04mm error would not describe the behavior you originally reported.

sjasonsmith, 2020-11-30:
> That range is a little higher than I was expecting

I actually had a hard piece of filament stuck to my nozzle which was hitting the bed when probing. These are my results after heating up and removing that. A 0.01mm range is more what I was expecting to see, based on previous experience with similar hardware.

```
M48 P20
...
Mean: 0.009375 Min: 0.005 Max: 0.015 Range: 0.010
Standard Deviation: 0.002484
```

sjasonsmith, 2020-12-05:
I am not receiving any new responses on this, so I just don't see anything actionable to work on here.
The claim that there is no leveling occurring has been completely debunked in my testing. I will be closing this since I don't see any work to be done.

I fully believe there could be issues remaining, but we would need new issues reported with very specific instructions to reproduce them.

If you are still having issues I recommend discussing them with some members of the community to see if there may be an error in the way you are using the feature, or a configuration change that could help. I personally am often present on Discord.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-12-05 by sjasonsmith