MarlinFirmware/Marlin #19841: [BUG] Compile error: lv_fs_res_t' does not name a type; did you mean 'lv_res_t

simpat1zq, 2020-10-21:
### Bug Description
When I compile, I get the following error. 

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5419102/Configuration.zip)

### Steps to Reproduce

Using those configs with auto build marlin, I compile and I get that error. 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-10-22:
Running 20007, this is old. 
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

simpat1zq, 2020-10-22:
@ellensp Thanks for your help. I converted the config over to the bugfix, and I still get the same message.I've attached my config files that I used with bugfix. :

`In file included from Marlin\src\MarlinCore.cpp:68:0:
Marlin\src\lcd/extui/lib/mks_ui/tft_lvgl_configuration.h:51:8: error: 'lv_fs_res_t' does not name a type; did you mean 'lv_res_t'?
 extern lv_fs_res_t spi_flash_open_cb (lv_fs_drv_t * drv, void * file_p, const char * path, lv_fs_mode_t mode);
        ^~~~~~~~~~~
        lv_res_t
Marlin\src\lcd/extui/lib/mks_ui/tft_lvgl_configuration.h:52:8: error: 'lv_fs_res_t' does not name a type; did you mean 'lv_res_t'?
 extern lv_fs_res_t spi_flash_close_cb (lv_fs_drv_t * drv, void * file_p);
        ^~~~~~~~~~~
        lv_res_t
Marlin\src\lcd/extui/lib/mks_ui/tft_lvgl_configuration.h:53:8: error: 'lv_fs_res_t' does not name a type; did you mean 'lv_res_t'?
 extern lv_fs_res_t spi_flash_read_cb (lv_fs_drv_t * drv, void * file_p, void * buf, uint32_t btr, uint32_t * br);
        ^~~~~~~~~~~
        lv_res_t
Marlin\src\lcd/extui/lib/mks_ui/tft_lvgl_configuration.h:54:8: error: 'lv_fs_res_t' does not name a type; did you mean 'lv_res_t'?
 extern lv_fs_res_t spi_flash_seek_cb(lv_fs_drv_t * drv, void * file_p, uint32_t pos);
        ^~~~~~~~~~~
        lv_res_t
Marlin\src\lcd/extui/lib/mks_ui/tft_lvgl_configuration.h:55:8: error: 'lv_fs_res_t' does not name a type; did you mean 'lv_res_t'?
 extern lv_fs_res_t spi_flash_tell_cb(lv_fs_drv_t * drv, void * file_p, uint32_t * pos_p);
        ^~~~~~~~~~~
        lv_res_t
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mks_robin_nano35\src\src\MarlinCore.cpp.o] Error 1
================================================= [FAILED] Took 15.75 seconds =================================================

Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:00:15.749
============================================ 1 failed, 0 succeeded in 00:00:15.749 ============================================ 
The terminal process "C:\Users\simpat\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
`

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5419590/Configuration.zip)

ellensp, 2020-10-22:
Your configuration wont even compile.
you have "define DOUBLECLICK_FOR_Z_BABYSTEPPING  // Double-click on the Status Screen for Z Babystepping." in Configuration_adv.h  wich should be #define.

simpat1zq, 2020-10-22:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5419657/Configuration_adv.zip)

Sorry, I uploaded the wrong version of the file. This one should give the error I mentioned.

ellensp, 2020-10-22:
nope. cannot reproduce your error.

simpat1zq, 2020-10-22:
Are you saying that it compiles with no error when you use those config files on the latest bugfix?

ellensp, 2020-10-22:
No it has errors, most  related to advanced_pause. but cannot replicate your errors

ellensp, 2020-10-22:
What did you change in platformio.ini ?

ellensp, 2020-10-22:
Regarding sdsupport. 

Talking with one of the higher up developers who does the TFT code.
"LVGL ui will not work without SDSUPPORT because it needs to loads the assets."

So SDSUPPORT is a requirement.

simpat1zq, 2020-10-22:
OK. I think the problem was with the platform.io, sorry about that. I was using a platform.io that someone else setup for this board, and now that I look at it, it's missing a bunch of stuff. I'm not sure how they got it to compile. I'm still getting compile errors, but I don't think it's a 'bug'.

closed 2020-10-22 by simpat1zq