MarlinFirmware/Marlin #19845: [BUG] new CREALITY_TOUCH not deploying

gbkwiatt, 2020-10-22:
### Bug Description

I have BLTouch from creality, and noticed new option for it CREALITY_TOUCH instead of usual BLTOUCH
However, when I go to configuration and BLTouch - deplying does not work (also when homing it's not deploying either)
Self test works 
My BLTouch is on pin 17

 

### My Configurations
ENDER 3 v427 with BLtouch connected to BLtouch connector on pin 17
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5421672/Marlin.zip)

Here is my configuration, on that file it's actually set to BLTOUCH instead of CREALITY_TOUCH but rest was the same for the test

### Steps to Reproduce

1. Enable CREALITY_TOUCH instead BLTOUCH
2. Go to LCD configuration screen -> BLtouch and press deploy

**Expected behavior:** it should deploy bltouch

**Actual behavior:** it doesn't deploy bltouch

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

xsuper9988, 2020-10-22:
is #19833 the same problem as this one?

gbkwiatt, 2020-10-22:
No, not at all, mine is not even deploying for homing or leveling

qwewer0, 2020-10-22:
BLTouch from Creality is still BLtouch, and not a CREALITY_TOUCH, that is for Creality CR-6 SE

gbkwiatt, 2020-10-22:
There you go,  thanks

closed 2020-10-22 by gbkwiatt

Sebazzz, 2020-10-25:
@qwewer0 So the CREALITY_TOUCH is actually the strain gauge?

qwewer0, 2020-10-25:
@Sebazzz As far as I know, yes.

Sebazzz, 2020-10-25:
> 
> 
> @Sebazzz As far as I know, yes.

Appears not to be though. CREALITY_TOUCH is checked, then defines BLTOUCH.

thinkyhead, 2020-10-25:
That option is for some probe that Creality used in developing Ender 3 V2 but it is not yet a real product.