MarlinFirmware/Marlin #19849: [Question] Chopper default on X2, Y2, Z2->4.

GMagician, 2020-10-22:
I was updating my cnf files when I found this, all above steppers, if chopper is not overriden, use `CHOPPER_TIMING_E` is it right? or is a copy&paste error?

```c++
#ifndef CHOPPER_TIMING_X2
  #define CHOPPER_TIMING_X2 CHOPPER_TIMING_E
#endif
```

qwewer0, 2020-10-22:
I see that too:
CHOPPER_TIMING_X2 CHOPPER_TIMING_E
CHOPPER_TIMING_Y2 CHOPPER_TIMING_E
CHOPPER_TIMING_Z2 CHOPPER_TIMING_E
CHOPPER_TIMING_Z3 CHOPPER_TIMING_E
CHOPPER_TIMING_Z4 CHOPPER_TIMING_E

Fix for that https://github.com/MarlinFirmware/Marlin/pull/19850

boelle, 2020-10-22:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

qwewer0, 2020-10-22:
@boelle I think there is still a bug, even if the title say [Question]...

boelle, 2020-10-22:
@GMagician will #19850 fix the issue?

boelle, 2020-10-22:
and please edit the title so it starts with [BUG] and not [QUESTION]

closed 2020-10-23 by thinkyhead

GMagician, 2020-10-23:
> @GMagician will #19850 fix the issue?

@boelle that's not related to my configuration, is just something I noticed.