MarlinFirmware/Marlin #19851: # include_next <stdint.h>

ptmrankzy, 2020-10-22:
I try to build the ender 3 pro v427 firmware but when its building it gives me an error.

files: [Ender3-pro Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5425985/Ender3-pro.Marlin.zip)

terminal:
`PS C:\Users\laszl\Documents\Ender3-pro Marlin> platformio run -e STM32F103RET6_creality
Processing STM32F103RET6_creality (platform: ststm32@~6.1; board: genericSTM32F103RE; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\users\laszl\documents\ender3-pro marlin\marlin\src\inc\MarlinConfigPre.h:31:0,
                 from c:\users\laszl\documents\ender3-pro marlin\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
c:\users\laszl\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\include\stdint.h:9:16: fatal error: stdint.h: No such file or directory

****************************************************************
* Looking for stdint.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:stdint.h"
* Web  > https://platformio.org/lib/search?query=header:stdint.h
*
****************************************************************

 # include_next <stdint.h>
                ^~~~~~~~~~
compilation terminated.
CalledProcessError: Command 'arm-none-eabi-g++.exe -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DSTM32_XL_DENSITY -DTEMP_TIMER_CHAN=4 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\laszl\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(item, exports="env")
  File "C:\Users\laszl\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\laszl\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\laszl\Documents\Ender3-pro Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 286:
    apply_features_config()
  File "C:\Users\laszl\Documents\Ender3-pro Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\laszl\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\laszl\Documents\Ender3-pro Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 262:
    load_marlin_features()
  File "C:\Users\laszl\Documents\Ender3-pro Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\laszl\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\laszl\.platformio\python3\lib\subprocess.py", line 512:
====================================================================================================== [FAILED] Took 0.81 seconds ====================================================================================================== 

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:00.810
================================================================================================= 1 failed, 0 succeeded in 00:00:00.810 =================================================================================================PS C:\Users\laszl\Documents\Ender3-pro Marlin> echo "done" >C:\Users\laszl\AppData\Local\Temp\ipc
PS C:\Users\laszl\Documents\Ender3-pro Marlin> `

boelle, 2020-10-22:
is this  [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

boelle, 2020-10-22:
in the zip /marlin/readme.md it says:

```# Ender 3 Pro "V1.5"

This is a surprise upgrade to the Ender 3 Pro that some customers began to receive around mid-July of 2020. No documentation or support page exists for this variant. It appears to be an Ender 3 Pro with the 32-bit Ender 3 V2 board and the stock Ender 3 Pro display. To see which version you have, examine the control board. The newer board is Creality v4.2.2.

This configuration is very similar to the Ender 3 V2 config except that the CR-10 stock display is enabled.```

that file is not present in https://github.com/MarlinFirmware/Marlin/tree/2.0.x/Marlin

So this almost seems like you got a copy of marlin elsewhere

ptmrankzy, 2020-10-22:
It is the bugfix branch

shitcreek, 2020-10-22:
`c:\users\laszl.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\include\stdint.h:9:16: fatal error: stdint.h: No such file or directory`

Looks like you are missing a PlatformIO Library. Maybe you need to update or reinstall PlatformIO, but I'd try searching/asking over on https://community.platformio.org/ first.

This isn't a Marlin issue I don't think.

boelle, 2020-10-22:
> It is the bugfix branch

so where did this file come from?:

```
# Ender 3 Pro "V1.5"

This is a surprise upgrade to the Ender 3 Pro that some customers began to receive around mid-July of 2020. No documentation or support page exists for this variant. It appears to be an Ender 3 Pro with the 32-bit Ender 3 V2 board and the stock Ender 3 Pro display. To see which version you have, examine the control board. The newer board is Creality v4.2.2.

This configuration is very similar to the Ender 3 V2 config except that the CR-10 stock display is enabled.
```

ellensp, 2020-10-22:
the readme was part of the PR, but not committed https://github.com/MarlinFirmware/Configurations/pull/161/files

ellensp, 2020-10-22:
Nb "asking since marlin does not have a v427 firmware" is incorrect I added 4 examples of Ender machines with this controller 19 days ago.

ellensp, 2020-10-22:
Using provided configs on marlin current bugfix compiles, with two warnings.

ellensp, 2020-10-22:
See if there are any differences from my build environment
```
> Executing task: platformio run <

Processing STM32F103RET6_creality (platform: ststm32@~6.1; board: genericSTM32F103RE; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RE.html
PLATFORM: ST STM32 (6.1.1) > STM32F103RE (64k RAM. 512k Flash)
HARDWARE: STM32F103RET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <U8glib-HAL> 0.4.3
|   |-- <Wire> 1.0
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
```

closed 2020-10-23 by thinkyhead