MarlinFirmware/Marlin #19854: [BUG] Marlin loses control of hotend during long print

rq3, 2020-10-22:
During a 7 hour print under 2.0.7.2, the printer consistently loses PID control of the hotend temperature at hour 5. This was repeatable over 3 prints. At the fifth hour (300 minutes, irrespective of layer or other factors), the hotend temperature enters a slow yet increasing oscillation mode until the printer freezes with an E1 temperature error (hotend temperature in excess of 10C above or below set point).

Please don't tell me to do a PID check, or verify that the sensor is OK. Been there. Done that. Modifying configuration_adv.h to allow this is also unacceptable, as a 20 degree swing in nozzle temperature is not OK during any print.

Anycubic Predator with Trigorilla Pro board.

Update: Confirmed during another print. Print fails at 4 hours, 16 minutes (256 minutes, 2^8 minutes). Files attached.

These files should be the "example" configurations for the stock Anycubic Delta Predator, since the ones that are on the Marlin GitHub are for an SK modified machine running Klipper, always have been, and are not useful for anyone attempting to run a stock machine using the Trigorilla Pro board and its stock LCD display with touch screen.
[2072_Stock_Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5426516/2072_Stock_Configuration.zip)

thisiskeithb, 2020-10-23:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19826

thisiskeithb, 2020-10-23:
Please don't create multiple issues for the same thing.

closed 2020-10-23 by thisiskeithb