MarlinFirmware/Marlin #19857: [BUG] Incorrect fan output?

Borillion, 2020-10-23:
### Bug Description

Incorrect fans turning on when commands are issues. 
SKR Pro 1.1 here I seem to have discovered a bug, or there is something I am being dense about.

### My Configurations

**Configuration_adv.h**
#define E0_AUTO_FAN_PIN PE5
....
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED 255   // 255 == full speed

**pins_BTT_SKR_PRO_common.h**
#define FAN_PIN                             PC8   // Fan0 (Part Cooling)
#define FAN1_PIN                            PE5   // Fan1 (Hotend Cooling)
#define FAN2_PIN                            PE6

//turn on hotend fan with at 50c
#ifndef E0_AUTO_FAN_PIN
  #define E0_AUTO_FAN_PIN               FAN1_PIN
#endif

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5426756/Configuration.zip)

### Steps to Reproduce

Issue M106 P0 S225, hotend cooling fan spins up.
Set hotend temp to 55C, part cooling fan spins up. 

**Expected behavior:**
Issue M106 P0 S225, part cooling fan should spin up.
Set hotend temp to 55C, part hotend fan should spin up. 

**Actual behavior:**
Issue M106 P0 S225, hotend cooling fan spins up.
Set hotend temp to 55C, part cooling fan spins up. 

[image]
[image]

#### Additional Information

I was able to fix this issue by swapping the connections, but this does not match the pin diagram provided by Bigtree Technology Co.
[image]

CaezarCZ, 2020-10-24:
I observed same problem

sjasonsmith, 2020-10-31:
I cannot reproduce this issue. I am using your Configuration and pins file with commit a4a487b0cc0a431c01ac07c8170b8a5f19cffd2e.

My board is not in a printer, so I am using the LED next to each fan to determine which are turned on.
I did connector a hotend and thermistor to observe auto-fan behavior.

Please try with the very latest state of the `bugfix-2.0.x` branch. I am not aware of recent changes, but something may have been fixed prior to my testing.

If you still have problems. Please look at the LED next to the fan connectors to verify which turns on with each fan. I was able to manually controll Fan0 and Fan2, with Fan1 (the middle fan) being automatically controlled by the heater.

sjasonsmith, 2020-11-16:
Two users seemingly had this problem, but neither have responded to my last post in over two weeks.
I am closing this, since it is working fine according to my testing.

If you have additional information based on the latest state of `bugfix-2.0.x`, please reply and this can potentially be re-opened.

closed 2020-11-16 by sjasonsmith