MarlinFirmware/Marlin #19858: Ender 3 v2 with BTT SKR Mini E3 V2

Stalkeros000, 2020-10-23:
Hi all. I am trying to connect the board to this printer, used different configs for Marlin (Bigtreetech and Marlin), but everywhere I run into a problem.
If I specify board "default_envs = STM32F103RC" in the file platformio.ini (I also tried boards STM32F103RC_btt; STM32F103RC_BTT_512K), and in the file Configuration.h specify:
 #ifndef MOTHERBOARD
define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0, 

then an error occurs during compilation:

```
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\dwin.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\rotary_encoder.cpp.o] Error 1
```

[image]
[https://ibb.co/ydSn2Hx](screen)
Please give advice. Thanks.

<details><summary>Log Output</summary>

```
. . .
Marlin\src\lcd\dwin\e3v2\dwin.cpp:2263:5: error: 'buzzer' was not declared in this scope
     buzzer.tone(100, 659);
     ^~~~~~
Marlin\src\lcd\dwin\e3v2\dwin.cpp:2263:5: note: suggested alternative: 'bzero'
     buzzer.tone(100, 659);
     ^~~~~~
     bzero
Marlin\src\lcd\dwin\e3v2\dwin.cpp:2268:5: error: 'buzzer' was not declared in this scope
     buzzer.tone(40, 440);
     ^~~~~~
Marlin\src\lcd\dwin\e3v2\dwin.cpp:2268:5: note: suggested alternative: 'bzero'
     buzzer.tone(40, 440);
     ^~~~~~
     bzero
Compiling .pio\build\STM32F103RC_btt\src\src\lcd\fontutils.cpp.o
In file included from Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/HAL.h:36:0,
                 from Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/HAL.h:26,
                 from Marlin\src\lcd\dwin\e3v2\../../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\dwin\e3v2\rotary_encoder.h:32,
                 from Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:35:
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp: In function 'void Encoder_tick()':
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:50:9: error: 'BEEPER_PIN' was not declared in this scope
   WRITE(BEEPER_PIN, 1);
         ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = (1U << PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:50:9: note: suggested alternative: 'TEMP_7_PIN'
   WRITE(BEEPER_PIN, 1);
         ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:33:42: note: in definition of macro 'WRITE'
 #define WRITE(IO,V)             (PIN_MAP[IO].gpio_device->regs->BSRR = (1U << PIN_MAP[IO].gpio_bit) << ((V) ? 0 : 16))
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp: In function 'ENCODER_DiffState Encoder_ReceiveAnalyze()':
Marlin\src\lcd\dwin\e3v2\rotary_encoder.h:44:34: error: 'BTN_EN1' was not declared in this scope
 #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
                                  ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:32:42: note: in definition of macro 'READ'
 #define READ(IO)                (PIN_MAP[IO].gpio_device->regs->IDR & (1U << PIN_MAP[IO].gpio_bit) ? HIGH : LOW)
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:79:7: note: in expansion of macro 'BUTTON_PRESSED'
   if (BUTTON_PRESSED(EN1)) newbutton |= 0x01;
       ^~~~~~~~~~~~~~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.h:44:34: error: 'BTN_EN2' was not declared in this scope
 #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
                                  ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:32:42: note: in definition of macro 'READ'
 #define READ(IO)                (PIN_MAP[IO].gpio_device->regs->IDR & (1U << PIN_MAP[IO].gpio_bit) ? HIGH : LOW)
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:80:7: note: in expansion of macro 'BUTTON_PRESSED'
   if (BUTTON_PRESSED(EN2)) newbutton |= 0x02;
       ^~~~~~~~~~~~~~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.h:44:34: error: 'BTN_ENC' was not declared in this scope
 #define BUTTON_PRESSED(BN) !READ(BTN_## BN)
                                  ^
Marlin\src\lcd\dwin\e3v2\../../../inc/../HAL/./STM32F1/fastio.h:32:42: note: in definition of macro 'READ'
 #define READ(IO)                (PIN_MAP[IO].gpio_device->regs->IDR & (1U << PIN_MAP[IO].gpio_bit) ? HIGH : LOW)
                                          ^~
Marlin\src\lcd\dwin\e3v2\rotary_encoder.cpp:81:7: note: in expansion of macro 'BUTTON_PRESSED'
   if (BUTTON_PRESSED(ENC)) {
       ^~~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\dwin.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt\src\src\lcd\dwin\e3v2\rotary_encoder.cpp.o] Error 1
======================================================= [FAILED] Took 88.27 seconds =======================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  FAILED    00:01:28.271
================================================== 1 failed, 0 succeeded in 00:01:28.271 ==================================================
The terminal process "C:\Users\Stalker\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
</details>

jernejp21, 2020-11-02:
@thinkyhead I am using this configuration on my printer. I have changed some code in `pins_BTT_SKR_MINI_E3_common.h` to compensate for different pinout. Will you except update if I make a pull request that only adds code for DWIN display and different pinout? User will have to make an adapter or new ribbon cable.

sjasonsmith, 2020-11-02:
> @thinkyhead I am using this configuration on my printer. I have changed some code in `pins_BTT_SKR_MINI_E3_common.h` to compensate for different pinout. Will you except update if I make a pull [...]

Personally, I do not think that displays should be added to pins files if they cannot be used without custom cabling. Can you not implement this purely through pin mapping? I assume you don't have serial pins where you need them?

sjasonsmith, 2020-11-02:
I suppose E3 V2 displays on E3-specific boards is kind of a special case. Maybe add the code, but comment it out and include an `#error`? Comments nearby provide a description of the hacks needed to make it work?

jernejp21, 2020-11-02:
TX, RX and reset pins are on different positions. Serial communication can be done in SW, but you cannot disable reset pin on STM32 like you can on AVRs.

sjasonsmith, 2020-11-02:
> Serial communication can be done in SW

SoftwareSerial is inadequate to control an LCD or to print over. It needs real serial pins.