MarlinFirmware/Marlin #19859: [BUG] Marlin 2.0.7.2 + MKS Robin nano v1.2 - UI is no longer properly scaled

dburr, 2020-10-23:
### Bug Description

Upgrading to Marlin 2.0.7.2 on a SainSmart Coreception 300 (MKS Robin nano v1.2 + MKS TFT). The LCD is no longer scaled properly. (in fact it appears as if the UI is not scaled at all - it is shoved up into the upper left corner)

A picture is worth a thousand words:
[image]

### My Configurations

[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5426921/conf.zip)

### Steps to Reproduce

1. Check out fresh 2.0.7.2 tree, copy in above configuration
2. Build

**Expected behavior:** Normally scaled screen

**Actual behavior:** Improperly scaled screen

#### Additional Information

Note that, although the screen itself is not correctly scaled, the touch targets are. (i.e. tapping on the "x" button where it is actually shown on screen does nothing, but if I tap in the lower left corner of the screen, where the button would have been if the screen is properly scaled, does do something.)

This exact same configuration, when built under 2.0.7, produces a Marlin build with a correctly scaled screen.

rhapsodyv, 2020-10-23:
Try bugfix. All TFT were updated.

dburr, 2020-10-23:
OK, turns out that I was supposed to `#define MKS_ROBIN_TFT35` to set my TFT type. This was not required in 2.0.7, however the `CONFIGURATION_H_VERSION` was not bumped (one would think that adding this new requirement would necessitate a config version bump.) Anyways all is well now, closing this PR.

closed 2020-10-23 by dburr