MarlinFirmware/Marlin #19861: [BUG] ultralcd_HD44780.cpp calling private function

TijsP, 2020-10-23:
### Bug Description

When selecting RA_CONTROL_PANEL as your LCD screen, an error gets thrown upon build.
```
.pio\libdeps\mega2560\LiquidCrystal/LiquidCrystal_I2C.h:186:9: error: 'int LiquidCrystal_I2C::init()' is private
    int  init();
         ^
Marlin\src\lcd\HD44780\ultralcd_HD44780.cpp:357:14: error: within this context
     lcd.init();
              ^
```
It seems like RA_CONTROL_PANEL defines LCD_I2C_TYPE_PCA8574 in Conditionals_LCD.h, which leads to lcd.init() being called in ultralcd_HD44780.cpp even though lcd.init() is a private function.
My best bet would be that instead of init(), begin() should be called instead.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5428252/Configuration.zip)

### Steps to Reproduce

1. Define RA_CONTROL_PANEL in Configuration.h
2. Build Marlin

**Expected behavior:** A successful build

**Actual behavior:** A failed build

**Additional information**

While this is likely an error every build target will get, I did try building for multiple boards (mega1280, mega2560, sanguino1284p and STM32F103RC), all of which resulted in the same error

ellensp, 2020-10-23:
In Marlin/src/lcd/HD44780/marlinui_HD44780.cpp find
```
#elif ENABLED(LCD_I2C_TYPE_PCA8574)
  LCD_CLASS lcd(LCD_I2C_ADDRESS, LCD_WIDTH, LCD_HEIGHT);
#elif ENABLED(SR_LCD_2W_NL)
```
change to
```
#elif ENABLED(LCD_I2C_TYPE_PCA8574)
  LCD_CLASS lcd(LCD_I2C_ADDRESS);
#elif ENABLED(SR_LCD_2W_NL)
```
and find
```
  #elif ENABLED(LCD_I2C_TYPE_PCA8574)
    lcd.init();
```
and change to
```
  #elif ENABLED(LCD_I2C_TYPE_PCA8574)
    lcd.begin(LCD_WIDTH,LCD_HEIGHT);
```

This compiles, can you test? I don't have this hardware

ellensp, 2020-10-23:
Does the lcd actually have a backlight?  the URL to this display is long dead. and wayback machine doesn't have the data.

TijsP, 2020-10-23:
To answer your second question first: I took a quick look at the datasheet of the PCA8574, and it doesn't look like it has a backlight feature

I replaced the lines like you suggested, and while it does upload successfully, the LCD isn't behaving properly. The first and third line are filled with black squares and the second and fourth are complete blank. I'll take a further look myself later this day, I'm a bit short on time right now. Still, help/some pointers would be greatly appreciated

ellensp, 2020-10-23:
That was a red herring...

The error message you saw was because marlin is including the wrong library
If you force it to use the correct library, the code compiles as it was. 

Please test this
In platformio.ini  change the mega2560 env: to this
```
[env:mega2560]
platform            = atmelavr
extends             = common_avr8
board               = megaatmega2560
lib_ignore          = LiquidCrystal
lib_deps            = ${common.lib_deps}
  LiquidCrystal_I2C
```

ellensp, 2020-10-23:
how to do this automatically
In platformio.ini find the line
`USES_LIQUIDCRYSTAL      = LiquidCrystal@1.5.0`
add this line under it
`USES_LIQUIDCRYSTAL_I2C  = LiquidCrystal_I2C@1.1.4`

In Marlin/buildroot/share/PlatformIO/scripts/common-dependencies.h find
`#elif ANY(HAS_MARLINUI_HD44780, LCD_I2C_TYPE_PCF8575, LCD_I2C_TYPE_PCA8574, SR_LCD_2W_NL, LCM1602)`
change to
```
#elif ENABLED(LCD_I2C_TYPE_PCA8574)
  #define USES_LIQUIDCRYSTAL_I2C
#elif ANY(HAS_MARLINUI_HD44780, LCD_I2C_TYPE_PCF8575, SR_LCD_2W_NL , LCM1602)
```

TijsP, 2020-10-24:
That works perfectly! I'm new to PlatformIO and VScode, so i definitely wouldn't have thought to check that. Thanks so much for your help!

closed 2020-10-24 by TijsP

ellensp, 2020-10-24:
Ill turn this in a PR so its fixed for all.