MarlinFirmware/Marlin #19862: REPRAP DISCOUNT SMART CONTROLLER Displaying 20x2 not 20x4

osherratt12, 2020-10-23:
### Bug Description

Have just upgraded to a BTT SKR 1.4 Turbo and plugged in my REPRAP DISCOUNT SMART CONTROLLER. It it only display information on the top 2 rows.

Have plugged the LCD back in to the old motherboard to check the LCD isn't broken and it still works fine.

It looks like it thinks the LCD is a 20x2 not a 20x4.

### My Configurations

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5428606/Archive.zip)

#### Additional Information

[image]

osherratt12, 2020-10-23:
Have fixed the issue

Added LCD_HEIGHT and LCD_WIDTH to conditionals_LCD.h 

```
#elif ENABLED(REPRAP_DISCOUNT_SMART_CONTROLLER)

  // RepRapDiscount LCD or Graphical LCD with rotary click encoder
  #define LCD_HEIGHT 4
  #define LCD_WIDTH  20
  #define IS_RRD_SC  1

#endif
```

[Conditionals_LCD.h.zip](https://github.com/MarlinFirmware/Marlin/files/5429027/Conditionals_LCD.h.zip)

ellensp, 2020-10-23:
Are you going to create a PR?

osherratt12, 2020-10-23:
PR created 

#19864

thisiskeithb, 2020-10-23:
Can you try with the latest `bugfix-2.0.x`? Your configs are missing several of the latest updates.

I flashed my SKR 1.4 and 1.3 builds with a `REPRAP_DISCOUNT_SMART_CONTROLLER` and they work fine.

osherratt12, 2020-10-23:
Have just downloaded the latest build and flashed and is now working. Very weird.

closed 2020-10-23 by osherratt12

sjasonsmith, 2020-10-23:
I probably patched this a week or so ago. I happened upon a warning related to it, but you would only noticed if you enabled the Prusa display style.