MarlinFirmware/Marlin #19863: [FR] Slim MarlinUI

ETE-Design, 2020-10-23:
For people with older mainboards It's not possible to use MarlinUI cause the low memory on the boards.
So mabye a Slim MarlinUI like the Slim Menu LCD could be the way to go to offer MarlinUI to even more mainboards.

thisiskeithb, 2020-10-23:
The quickest way to a smaller build in the meantime is to disable features you don't need. Searching the config files for `PROGMEM` will point you to some heavier items that can help shrink a build down.

ETE-Design, 2020-10-23:
@thisiskeithb How much does the MarlinUI require of PROGMEM compared to Slim LCD Menu?

thisiskeithb, 2020-10-23:
> @thisiskeithb How much does the MarlinUI require of PROGMEM compared to Slim LCD Menu?

Both are still Marlin UI, but `SLIM_LCD_MENUS` just leaves out extraneous menus. You'll have to compile your specific build and test the difference with your required options enabled.

ETE-Design, 2020-10-24:
@thisiskeithb  Sorry, forgot to mention that it is the MarlinUI for TFT I was talking about.

thisiskeithb, 2020-10-24:
> Sorry, forgot to mention that it is the MarlinUI for TFT I was talking about.

There are three:

https://github.com/MarlinFirmware/Marlin/blob/9630c2683c39468bf885c560a1768d8b377d9462/Marlin/Configuration.h#L2300-L2302

`TFT_CLASSIC_UI` just emulates the standard Marlin UI.

I think what you're after is shrinking down all the TFT code down to a smaller size, not the `TFT_CLASSIC_UI`. The TFT code is really meant for a 256k chip at the very least.

ETE-Design, 2020-10-24:
@thisiskeithb So the TFT_Classic_UI can also emulator Slim LCD Menu? But yes in general I was thinking about cutting the TFT Code down to as small as possible, like you slim total side down in LCD when you choose Slim LCD Menu...

thisiskeithb, 2020-10-24:
I haven't tried `TFT_CLASSIC_UI` and `SLIM_LCD_MENUS`, so I can't speak to the final size/if those are compatible, but I don't see why they wouldn't be.

The TFT UI code will already be small because it won't compile all three options.

thisiskeithb, 2021-12-02:
Closing because it’s not possible to create a “slim” version of any of the TFT UIs.

They may be optimized and shrink the overall build size down the road, but they aren’t meant for small MCUs.

closed 2021-12-02 by thisiskeithb