MarlinFirmware/Marlin #19866: [BUG] HOMING_FEEDRATE_XY to fast for ender 3 v2?

pixelicous, 2020-10-23:
Not sure if this is a bug
According to #12074, HOMING_FEEDRATE_XY was supposed to be changed back from 50*60 to 20*60
It seems like in ender 3 v2 the configuration on bugfix 2.x is back to 50*60.

Is this intentional or not??

ellensp, 2020-10-23:
Ender 3 and Ender 3 v2 are different machines, years apart...  Can you confirm this is a issues on a real V2?

pixelicous, 2020-10-23:
@ellensp I know but kind of figured these hardware stop switches didn't change much even though the board did..

Once my current print finished i will check it out.. but i was just wondering if anybody actually tested that 50/60 is ok on ender 3 v2 compared to v1? or just decided to put it in..

again will test shortly and reply again

thanks!

thisiskeithb, 2020-10-23:
Ender-3 V2 support was added by @thinkyhead who has a real machine to test code on.

pixelicous, 2020-10-23:
@ellensp @thisiskeithb 
Well i'm sure he did check it, but I just looked, if I home my printer, the little metal tabbed is pressed all the way in, exactly like in the original post of ender 3.  I will go ahead and compile it with 20/60 and post if there is any difference..

pixelicous, 2020-10-24:
Well i compiled it but the result somehow was the same, in terms of that lever, not speed of course.
Checked creality's source config for their own complied files, they have the same 50*60, so i guess its safe although it shows that level pressed down all the way :)

closed 2020-10-24 by pixelicous