MarlinFirmware/Marlin #19867: compilation error

Malderin, 2020-10-23:
Board: MKS robin nano v2.
After update https://github.com/MarlinFirmware/Marlin/commit/4533a197d59ffcaccc4d476f06df7754b11b9b0c (i think this) atom throws compilation error:
In file included from .pio\libdeps\mks_robin_nano35\lvgl/lvgl.h:41:0,
                 from Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:24,
                 from Marlin\src\lcd\extui\lib\mks_ui\draw_motor_settings.cpp:26:
.pio\libdeps\mks_robin_nano35\lvgl/src/lv_objx/lv_btn.h:120:12: note: candidate: lv_obj_t* lv_btn_create(lv_obj_t*, const lv_obj_t*)
 lv_obj_t * lv_btn_create(lv_obj_t * par, const lv_obj_t * copy);
            ^~~~~~~~~~~~~
.pio\libdeps\mks_robin_nano35\lvgl/src/lv_objx/lv_btn.h:120:12: note:   candidate expects 2 arguments, 8 provided
In file included from Marlin\src\lcd\extui\lib\mks_ui\draw_motor_settings.cpp:26:0:
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:189:11: note: candidate: lv_obj_t* lv_btn_create(lv_obj_t*, lv_event_cb_t, int, lv_style_t*)
 lv_obj_t* lv_btn_create(lv_obj_t *par, lv_event_cb_t cb, const int id, lv_style_t *style);
           ^~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:189:11: note:   candidate expects 4 arguments, 8 provided
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:192:11: note: candidate: lv_obj_t* lv_btn_create(lv_obj_t*, lv_event_cb_t, int)
 lv_obj_t* lv_btn_create(lv_obj_t *par, lv_event_cb_t cb, const int id);
           ^~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:192:11: note:   candidate expects 3 arguments, 8 provided
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:195:11: note: candidate: lv_obj_t* lv_btn_create(lv_obj_t*, lv_coord_t, lv_coord_t, lv_coord_t, lv_coord_
t, lv_event_cb_t, int)
 lv_obj_t* lv_btn_create(lv_obj_t *par, lv_coord_t x, lv_coord_t y, lv_coord_t w, lv_coord_t h, lv_event_cb_t cb, const int id);
           ^~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\mks_ui\draw_ui.h:195:11: note:   candidate expects 7 arguments, 8 provided
*** [.pio\build\mks_robin_nano35\src\src\lcd\extui\lib\mks_ui\draw_motor_settings.cpp.o] Error 1

Malderin, 2020-10-23:
My configuration files
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5429625/conf.zip)

rhapsodyv, 2020-10-23:
In fact, the PR broken LVGL compilation when using TMC.

rhapsodyv, 2020-10-23:
Try #19869

Malderin, 2020-10-23:
> Try #19869

Error:

Marlin\src\gcode\feature\trinamic\M569.cpp: In instantiation of 'void tmc_say_stealth_status(TMC&) [with TMC = TMCMarlin<TMC2209Stepper, 'E', '0',
(AxisEnum)3>]':
Marlin\src\gcode\feature\trinamic\M569.cpp:147:5:   required from here
Marlin\src\gcode\feature\trinamic\M569.cpp:35:21: error: 'class TMCMarlin<TMC2209Stepper, 'E', '0', (AxisEnum)3>' has no member named 'get_stealthC
hop_status'; did you mean 'get_stealthChop'?
   serialprintPGM(st.get_stealthChop_status() ? PSTR("stealthChop") : PSTR("spreadCycle"));
                  ~~~^~~~~~~~~~~~~~~~~~~~~~
                  get_stealthChop
*** [.pioenvs\mks_robin_nano35\src\src\gcode\feature\trinamic\M569.cpp.o] Error 1

rhapsodyv, 2020-10-23:
Hum... Try pull again... When I commented here, I had forgotten a commit.

Malderin, 2020-10-23:
> Try #19869

Marlin\src\gcode\feature\trinamic\M569.cpp: In instantiation of 'void tmc_say_stealth_status(TMC&) [with TMC = TMCMarlin<TMC2209Stepper, 'X', '0',
(AxisEnum)0>]':
Marlin\src\gcode\feature\trinamic\M569.cpp:123:5:   required from here
Marlin\src\gcode\feature\trinamic\M569.cpp:35:21: error: 'class TMCMarlin<TMC2209Stepper, 'X', '0', (AxisEnum)0>' has no member named 'get_stealthC
hop_status'; did you mean 'get_stealthChop'?
   serialprintPGM(st.get_stealthChop_status() ? PSTR("stealthChop") : PSTR("spreadCycle"));
                  ~~~^~~~~~~~~~~~~~~~~~~~~~
                  get_stealthChop
Marlin\src\gcode\feature\trinamic\M569.cpp: In instantiation of 'void tmc_say_stealth_status(TMC&) [with TMC = TMCMarlin<TMC2209Stepper, 'Y', '0',
(AxisEnum)1>]':
Marlin\src\gcode\feature\trinamic\M569.cpp:129:5:   required from here
Marlin\src\gcode\feature\trinamic\M569.cpp:35:21: error: 'class TMCMarlin<TMC2209Stepper, 'Y', '0', (AxisEnum)1>' has no member named 'get_stealthC
hop_status'; did you mean 'get_stealthChop'?
   serialprintPGM(st.get_stealthChop_status() ? PSTR("stealthChop") : PSTR("spreadCycle"));
                  ~~~^~~~~~~~~~~~~~~~~~~~~~
                  get_stealthChop
Marlin\src\gcode\feature\trinamic\M569.cpp: In instantiation of 'void tmc_say_stealth_status(TMC&) [with TMC = TMCMarlin<TMC2209Stepper, 'Z', '0',
(AxisEnum)2>]':
Marlin\src\gcode\feature\trinamic\M569.cpp:135:5:   required from here
Marlin\src\gcode\feature\trinamic\M569.cpp:35:21: error: 'class TMCMarlin<TMC2209Stepper, 'Z', '0', (AxisEnum)2>' has no member named 'get_stealthC
hop_status'; did you mean 'get_stealthChop'?
   serialprintPGM(st.get_stealthChop_status() ? PSTR("stealthChop") : PSTR("spreadCycle"));
                  ~~~^~~~~~~~~~~~~~~~~~~~~~
                  get_stealthChop
Marlin\src\gcode\feature\trinamic\M569.cpp: In instantiation of 'void tmc_say_stealth_status(TMC&) [with TMC = TMCMarlin<TMC2209Stepper, 'E', '0',
(AxisEnum)3>]':
Marlin\src\gcode\feature\trinamic\M569.cpp:147:5:   required from here
Marlin\src\gcode\feature\trinamic\M569.cpp:35:21: error: 'class TMCMarlin<TMC2209Stepper, 'E', '0', (AxisEnum)3>' has no member named 'get_stealthC
hop_status'; did you mean 'get_stealthChop'?
   serialprintPGM(st.get_stealthChop_status() ? PSTR("stealthChop") : PSTR("spreadCycle"));
                  ~~~^~~~~~~~~~~~~~~~~~~~~~
                  get_stealthChop
*** [.pioenvs\mks_robin_nano35\src\src\gcode\feature\trinamic\M569.cpp.o] Error 1

rhapsodyv, 2020-10-23:
Did you change the configs or they are the same? Because I'm not getting this error.

rhapsodyv, 2020-10-23:
`get_stealthChop_status` don't exists anywhere... I guess your bugfix was outdated when you merged my PR.... You need fresh bugfix + my PR

Malderin, 2020-10-23:
> `get_stealthChop_status` don't exists anywhere... I guess your bugfix was outdated when you merged my PR.... You need fresh bugfix + my PR

thanks, did so, helped.

PR #19869 (2020-10-24): More MKS UI cleanup, optimization
### Description

Follow up for 19843 when TMC is enabled. Added a test for that too.

### Benefits

Fix #19867 

### Related Issues

#19867 
#19843
closed 2020-10-23 by Malderin