MarlinFirmware/Marlin #19868: ERROR Serial port pins (1) conflict with LCD pins!

Rubenazzzo, 2020-10-23:
good afternoon, sorry for my english.
I am trying to compile a skr_v1.3 board with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.
when I activate it I get the following error mentioned.
I try to found the problem in a pins configuration , and don´t found nothing.

[image]
[image]

[configuracion_skr_v1.3.zip](https://github.com/MarlinFirmware/Marlin/files/5429914/configuracion_skr_v1.3.zip)

thisiskeithb, 2020-10-23:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-23 by thisiskeithb

ellensp, 2020-10-23:
change #define SERIAL_PORT 1 to #define SERIAL_PORT -1

Rubenazzzo, 2020-10-23:
THANK YOU VERY MUCH, it is not the first skr board that I configure, it blinds me with other parameters and not with the basic serial one.
I thought it was a problem with marlin. It was not a support question, sorry for the problems caused