MarlinFirmware/Marlin #19870: [BUG] M25/M125 without P1 parameter should wait for M24

qwewer0, 2020-10-23:
### Bug Description

When printing from SD card and the print reaches `M25`/`M125` without `P1` parameter, the following happens:
1. Printer gets ready for pause and completes the rest of the moves that is already planned (LCD shows parking...)
2. Printer rises Z axis by `NOZZLE_PARK_POINT` `Z_raise`, but at leaste by `NOZZLE_PARK_Z_RAISE_MIN` (if defined), but not higher than `Z_MAX_POS`
3. Printer moves to `NOZZLE_PARK_POINT` X and Y
4. After the hotend is parked, the printer moves back to the saved XY position (`resume_position`, XY), without any wait, but with the wrong (parking) Z height, like how it happened for @GMagician in https://github.com/MarlinFirmware/Marlin/issues/11428#issue-346724055 (Aug 1 2018), but in my case it only continues the print for a few gcode lines, then it stops at that position (waiths for `M24` with the heaters on)
5. At this point, the LCD still shows parking..., so there isn't a way to resume the print from the LCD
6. By sending `M24` through serial connection, the printer shows **RESUME OPTIONS** on the LCD, then if **Continue** is selected, it moves to the originaly saved XY position and lowers Z axis to the saved Z position (`resume_position`, XYZ), then it skipps the already executed gcode lines that was printed on the wrong Z height, and continues the print.

Video of the problem: https://imgur.com/a/xdUa7Cy
17th sec -> M25 Parking
28th sec -> M24 received

With my limited understanding it seems that for `M25`/`M125` without `P1` parameter, none of the next options happen:
1. Doesn't pauses the print
2. Resumes immediately after it parked/paused, effectively not pausing
3. Parks/pauses and waits indefinitely until `M24` is sent to the printer

I think that the second option is the worst of the three, and because we have ability to use P1 parameter, then the third option seems to be the best, most logical option. (https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-711418985)

I believe that there is an issue, even if it isn't reproducible for everyone, like how it worked (like in third option) for @rhapsodyv https://github.com/MarlinFirmware/Marlin/issues/18191#issuecomment-711416738

### My Configurations

Ender 3
SKR Mini E3 v1.2
Latest Bugfix-2.0.x (0ffee29)
SD Card: 8GB, 128MB
default_envs = `STM32F103RC_btt`:
[Configuration_256k.zip](https://github.com/MarlinFirmware/Marlin/files/5430389/Configuration_256k.zip)
RAM:   [===       ]  26.7% (used 13104 bytes from 49152 bytes)
Flash: [========= ]  85.8% (used 224876 bytes from 262144 bytes)

default_envs = `STM32F103RC_btt_512K`:
[Configuration_512k.zip](https://github.com/MarlinFirmware/Marlin/files/5430388/Configuration_512k.zip)
RAM:   [===       ]  29.1% (used 14296 bytes from 49152 bytes)
Flash: [=====     ]  52.9% (used 277532 bytes from 524288 bytes)

Test file:
[Shape-Box_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/5430385/Shape-Box_gcode.zip)
[Shape-Box_3mf.zip](https://github.com/MarlinFirmware/Marlin/files/5430386/Shape-Box_3mf.zip)

Requirements:
```
NOZZLE_PARK_FEATURE
SDSUPPORT
ADVANCED_PAUSE_FEATURE
PARK_HEAD_ON_PAUSE
Printing from SD Card
```

### Steps to Reproduce

1. Connect to the printer (Pronterface, Octoprint etc.)
2. Start a gcode with M25 in it
3. Watch the parking sequence
4. Send M24 to the printer

**Expected behavior:** [What you expect to happen]
`M25`/`M125` without `P1` parameter should pause and park the hotend, then wait until it receives `M24`. (third option)

**Actual behavior:** [What actually happens]
`M25`/`M125` without `P1` parameter, the printer pauses and parks the hotend, then immediately moves back to the saved XY position (prior to parking), without Z position, and executes a few gcode lines, then stops at that position above the print, and wait for `M24`.
After it receives `M24` through serial connection, the hotend moves to the saved XY and Z position (prior to parking), and starts the print skipping the already executed gcode lines.

qwewer0, 2020-11-24:
Still an issue on the latest bugfix-2.0.x.

qwewer0, 2020-12-25:
Still an issue on the latest bugfix-2.0.x 4b860f1.

Bluelinegecko, 2021-01-05:
Still an issue on the latest bugfix-2.0.x.

qwewer0, 2021-01-05:
@Bluelinegecko Could you provide configuration files?

Bluelinegecko, 2021-01-05:
> @Bluelinegecko Could you provide configuration files?

Yeah here they are...

[https://github.com/Bluelinegecko/Bug-19870/blob/main/Configuration.h](url)
[https://github.com/Bluelinegecko/Bug-19870/blob/main/Configuration_adv.h](url)

I'm not sure about the original poster's configuration, but in looking at the code in M25.cpp if I am reading the code correctly, it looks like it is only responding to the P1 parameter  in the gcode if "HAS_LCD_MENU" or "EXTENSIBLE_UI" is defined, but as far as I can tell when the DWIN_CREALITY_LCD (used on the Ender3 V2) is selected as a controller neither of these are defined. Coding is not my strongpoint so I'm not 100% sure I'm interpreting that right though. If this is the case it might be the cause of the bug.

Bluelinegecko, 2021-01-20:
I did some further testing on this issue on an Ender3V2 the last few days using the Bugfix2.0 branch. I suspect what is happening is that the queue or buffer is not being fully emptied/paused before the pause is takes effect and the printer continues to processes the next line or two after reading the M25 in the g-code.

Since this seems to be related to Creality printers with the DWIN display, I initially thought this might be related to the DWIN specific if/else statement at https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/sd/M24_M25.cpp#L102 but when I tested it out after deleting lines 101 & 103 so that the firmware processed the ui.reset_status() function it made no difference.

This printer is using the stock Creality 4.2.2 board, but as the original poster is using the SKR Mini E3 v1.2 I don't believe it is board specific. I wonder if the faster 32-bit boards might just be processing commands too quickly and that the when the printer sets the initial pause flag at https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/gcode/sd/M24_M25.cpp#L90 the next line or two of g-code makes it into the queue before the board sees the pause flag and halts movements/commands.

I say this because several people like me have been having issues when placing a pause command in G-code for SD card printing using these printers. This is affecting the use of pause-at-layer-height scripts when using Cura, PrusaSlicer, etc to pause SD cards to place captive nuts, perform manual filament swaps etc. What happens is that the printer reads the pause command (whether it is a M25 or M125) but regardless of whether the firmware has parking enabled or the G-code script moves the nozzle to a parking position, the nozzle goes to the park position and then immediately returns to where it left off printing and performs the pause at that location. This makes it difficult or impossible to insert captive nuts, and/or can ruin prints from material oozing out of the nozzle during the pause or the extra material extruded when purging/priming the nozzle upon resuming printing.

The P1 parameter does not seem to always work with M125 or M25, and it appears to me that the M25 function doesn't even process the P1 parameter when printing from a SD card. It appears to me that the P1 parameter is only processed when calling M25 if the printer is running from a host computer.

As I wrote earlier, I suspect it is related to the commands in the buffer/queue being processed after the pause is called because of a test I just ran.

I ran a simple G-code in which I manually added the M25 command in between some movement commands as seen here

> G1 F9000 X10 Y210
> M300
> M25 P1

The printer moved to the park position (X10 Y210), beeped, then moved back to the printing position in the middle of the bed where it stopped and waited for the resume command.

But by simply changing the script to read 

>G1 F9000 X10 Y210
>M25 P1
>M300
>M300
>M300

The printer moved to the park position WHILE beeping three times, and waited there for the resume command. this shows that the printer processed the 3 lines of code AFTER the M25 P1 command before the pause took effect.

I'm not sure how to go about coding a fix for this but hopefully it will help one of you guys familiar with it to come up with a solution.

qwewer0, 2021-01-20:
> The P1 parameter does not seem to always work with M125 or M25, and it appears to me that the M25 function doesn't even process the P1 parameter when printing from a SD card. It appears to me that the P1 parameter is only processed when calling M25 if the printer is running from a host computer.

Not sure how many times did it not work for you, but for me `M25` or `M125` **with** `P1` parameter always worked so far.

Bluelinegecko, 2021-01-20:
> Not sure how many times did it not work for you, but for me `M25` or `M125` **with** `P1` parameter always worked so far.

Marlin doesn't have the commands programmed in yet to display the prompts for the advanced pause features on the Creality DWIN display, so I ended up disabling M125 on this printer as I had to blindly hit the encoder wheel to navigate through the absent purge more/resume menus. Regardless, every time I tried running M125 P1 or M25 P1, it still returned from the park position to the point it left of printing and then paused there. This makes it impossible to insert captive hardware as well as leads to a big mess when the advanced pause functions purged filament upon resuming printing. 

The printer pauses correctly it just processes the next few lines of G-code before the pause takes effect. I was able to confirm the firmware is processing a few more lines of g-code after it reads the pause command but before the pause is completed by placing the beep commands after the pause command.

qwewer0, 2021-01-20:
> Marlin doesn't have the commands programmed in yet to display the prompts for the advanced pause features on the Creality DWIN display

Got it.

> , so I ended up disabling M125 on this printer as I had to blindly hit the encoder wheel to navigate through the absent purge more/resume menus.

Disable `M125`?

> Regardless, every time I tried running M125 P1 or M25 P1, it still returned from the park position to the point it left of printing and then paused there. This makes it impossible to insert captive [...]

Do you have `HAS_LCD_MENU` defined? (don't define it manually) https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L479-L481
Without `HAS_LCD_MENU`, the `P` parameter won't do anything.

Bluelinegecko, 2021-01-20:
> Do you have `HAS_LCD_MENU` defined? (don't define it manually) https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/inc/Conditionals_LCD.h#L479-L481
> Without `HAS_LCD_MENU`, the `P` parameter won't do anything.

It doesn't appear to be defined when selecting the DWIN_CREALITY_LCD as the display. 

 Also, it appears the P1 parameter is never processed when calling an M25 unless you are connected to a host computer. If I am reading it right it will completely ignore P1 if processing a print from the SD card. So if the firmware reads M25 from an SD card print, and the advanced pause features in Configuration_adv.h are not defined, then "PARK_HEAD_ON_PAUSE" won't be defined and the M25 command won't be processed as an M125 where the P1 parameter might be detected.
 
 But I think you are missing my point. The pause **does** work. It pauses and waits for the user to click resume on the DWIN display. It is just it processes a few more lines of Gcode before stopping which can either cause it to move to the next print coordinates, or like in my workaround, play a few beeps.

Guilouz, 2021-01-30:
I have an issue when I use M25 command.

When M25 is sending, the hotend is parked but immediately starts again above the print (at the last position where the pause was made) and a purge is done above the print. My screen tell me if I want to purge more or restart the print.

It's working when I use M125 P1

qwewer0, 2021-01-30:
@Guilouz Could you attach your configuration files?

Guilouz, 2021-01-30:
> @Guilouz Could you attach your configuration files?

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5898634/Configurations.zip) for my CR10S Pro

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5898637/Configurations.zip) for my Sidewinder X1

Both have issue with M25 and ok with M125 P1

qwewer0, 2021-03-04:
I think https://github.com/MarlinFirmware/Marlin/issues/20969 is a bit more in depth, so I will close this. If needed, this can be opened later.

closed 2021-03-04 by qwewer0