MarlinFirmware/Marlin #19872: [BUG] Russian language support is gone

Malderin, 2020-10-23:
Board: MKS robin nano v2.
After update https://github.com/MarlinFirmware/Marlin/commit/4533a197d59ffcaccc4d476f06df7754b11b9b0c(i think this) russian language support is gone. The easiest way to make sure of this is to try to print an LCD from a SD card.
The print start dialog now looks like this:
[image]
[image]

rhapsodyv, 2020-10-24:
Can you test this PR? #19869

thisiskeithb, 2020-10-24:
Now that PR #19869 is merged, please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Malderin, 2020-10-24:
> Now that PR #19869 is merged, please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

The error remained, tested on the last bugfix from https://github.com/MarlinFirmware/Marlin/pull/19869
There is a new error. When entering the menu setting/config and beyond "leveling settings" or "advance settings", printer freezes and restarts.
See video:
https://yadi.sk/i/nx8dJp8CYrGjNg

rhapsodyv, 2020-10-24:
I fixed the menu problems here: #19879.

Abou the RU lang, it's working. But you need to manually set it again, because the settings could be reseted. Settings->Lang->Select Russian.

Malderin, 2020-10-24:
> I fixed the menu problems here: #19879.
> 
> Abou the RU lang, it's working. But you need to manually set it again, because the settings could be reseted. Settings->Lang->Select Russian.

The menu freeze issue is solved, thanks.
But the problem from the first post still persists

rhapsodyv, 2020-10-24:
Which problem?

Russian is working for me: 
[image]

Malderin, 2020-10-24:
> Which problem?
> 
> Russian is working for me:
> [image]

The top-level menu is (almost) no problem.
On your screen there should be a translation:
[image]
In tft_Language_ru.h: #define TITLE_READYPRINT_RU                 "готов к"
and 
#define TITLE_SET_RU                        "настройки"
Should be so: 
[image]
Now look at the photo in my first post. 
This is how it should be:
[image]
To see this you need to do this:
1. Write any gcode file to the memory card
2. Click on it
3. This dialog will appear.

Malderin, 2020-10-24:
Earlier I translated all the "settings / config" menus.
On the first 2 levels, the Russian translation has been preserved, on the third level, Russian letters are not displayed.
For instance:
The "Config"  and "Machine settings" is displayed normally, and inside the "acceleration" menu, where you need to enter values, the Russian letters have disappeared. 
The same goes for all menus:
1. Motor settings (and one next) is ok, inside third menu (Steps settings and other) - without russian;
2. Leveling settins - similar to p. 1;
3. Advance settings - similar to p. 1.

Malderin, 2020-10-24:
If needed, here is my actual build where I am testing:
https://github.com/Malderin/Marlin_2.0.x_official

rhapsodyv, 2020-10-24:
Ok, I got it. I will check what is happening.

rhapsodyv, 2020-10-24:
Ok! Fixed that.

PR #19879 (2020-10-24): LVGL UI cleanup follow up
### Description

Minor fixes after the cleanup of LVGL UI.

### Benefits

Fix #19872

### Related Issues

#19869 
#19843
closed 2020-10-24 by thisiskeithb