MarlinFirmware/Marlin #19873: pins_CHITU3D_V5.h and pins_CHITU3D_V6.h have FAN_PIN_2

ellensp, 2020-10-24:
### Bug Description

Noticed that pins_CHITU3D_V5.h and pins_CHITU3D_V6.h have FAN_PIN_2 defined.
But marlin doesn't use a naming convention of FAN_PIN_{n}
Marlin uses FAN{n}_PIN 

### My Configurations

In this rare case configs are not needed. 

### Steps to Reproduce

1. Open editor  
2. Open pins_CHITU3D_V5.h or pins_CHITU3D_V6.h 
3. Note FAN_PIN_2 is defined.

Additional info

A easy fix, I just don't have time to create a PR right now. So documented it.

thisiskeithb, 2020-10-24:
See https://github.com/MarlinFirmware/Marlin/pull/19874

The config will need to be updated as well: https://github.com/MarlinFirmware/Configurations/blob/568358466388dfc3a0b946d13962bbbe97477838/config/examples/Tronxy/X5SA/Configuration_adv.h#L477

ellensp, 2020-10-24:
you had do go and find a config that was effected :P good catch!

ellensp, 2020-10-24:
merged and closed.

closed 2020-10-24 by ellensp