MarlinFirmware/Marlin #19876: [BUG] FTDI touch screen not working with software SPI

MarkMan0, 2020-10-24:
### Bug Description

I'd like to use a VM800B35A display on my printer. Using hardware SPI it works correctly, but when I switch to Software SPI it doesn't "boot", just blank white screen. I have Debug enables, on the serial I'm getting:

```
Time to draw screen (ms): 10
Fault waiting for space in the command processor
Fault waiting for space in the command processor
echo:Appending to DL from RAMG cache, bytes: 65535 REG_CMD_DL: 4294967295
```
The Fault waiting for space in the command processor is spammed.

### My Configurations

I'm using a BTT SKR PRO v1.1
For hardware SPI I'm using the EXP2 pins for the SD communication(SDSUPPORT is disabled) - this is working correctly
For soft SPI, I'd like to use EXP1. in my[ Configuration_adv.h ](https://github.com/MarkMan0/Marlin/blob/extUI_bugfix/Marlin/Configuration_adv.h#L1497)you can find the pin definitions, AFAIK it's correct. Right now it's configured for hardware SPI
I've also tried to use the EXP2 pins in software SPI, without success.

The files are on github, but if required I can post a zip too.
https://github.com/MarkMan0/Marlin/tree/extUI_bugfix

### Steps to Reproduce

1. Check display with hardware SPI
2. configure software SPI
3. Check if display is still working

**Expected behavior:** 

Display working with both software and hardware SPI

**Actual behavior:**

Display only working with hardware SPI

#### Additional Information

Ultimately I'd like to use EXP1 for the display and EXP2 for the SD card. Could marlin handle both the SD and the display on the hardware SPI line, or are there race conditions?

I believe the issue is that the bit-banging is too fast in SPI::_soft_spi_xfer and others, I think some delays should be introduced.
 Unfortunately I don't have any experience. If someone has any suggestions on where and how much delay to add, I'm happy to try them.

MarkMan0, 2020-10-24:
I've tried adding using DELAY_NS, like this:
```
      for (k = 0; k < 8; k++) {  // Output and Read each bit of spiOutByte and spiInByte
        WRITE(CLCD_SOFT_SPI_MOSI, (spiOutByte & spiIndex) ? 1 : 0); // Output MOSI Bit
        WRITE(CLCD_SOFT_SPI_SCLK, 1);   // Pulse Clock
        DELAY_NS(1000);
        if (READ(CLCD_SOFT_SPI_MISO)) spiInByte |= spiIndex; // MISO changes on the falling edge of clock, so sample it before
        WRITE(CLCD_SOFT_SPI_SCLK, 0);
        DELAY_NS(1000);
        spiIndex >>= 1;
      }
```

I've tried higher and lower values, but I always get `Waiting for touch release`, and the the board resets - I assume the watchdog times out

rhapsodyv, 2020-11-30:
Why do you need FTDI with Software SPI? Any specific reason?

I'm asking that because soon we will use sw spi only on specific cases and when needed, everything else will run on hw spi (when available).

> Ultimately I'd like to use EXP1 for the display and EXP2 for the SD card. Could marlin handle both the SD and the display on the hardware SPI line, or are there race conditions?

Yes. But not currently, I guess. We are working on some SPI fixes that will be released soon.

MarkMan0, 2020-11-30:
It's not that I can't work around it, but I found a bug and I reported it. I already switched to HW SPI, but without SD.

rhapsodyv, 2020-11-30:
One more question: FTDI is a TFT? Who many data goes through SPI?

I'm asking that, because SW SPI could be too slow to work on TFTs...

rhapsodyv, 2020-12-04:
@marciot do you want to take a look on this? Or SW SPI is something that is deprecated for FTDI?

Thanks!

marciot, 2021-01-03:
The software SPI functionality is maintained only for the Archim board and I know it is working fine there; fully supporting other boards is not something I have the the time or interest to investigate.

closed 2021-01-03 by rhapsodyv