MarlinFirmware/Marlin #19877: [BUG] M600 (Filemant Change) not working in filament runout script for mixing extruders

the-real-orca, 2020-10-24:
### Bug Description

Normally M600 falls back to the active extrude if no extrude / tool is specified with `Tx`. This works for single and (basic) multiple extruders, however once `MIXING_EXTRUDER` is enabled an *"E stepper not specified"* error is thrown if parameter `Tx` is omitted. 
I would have expected a similar behavior as for a basic dual extruder setup. Especially if you are using `M600` as the filament runout script.

### My Configurations

#### dual extrude setup -> OK

``` C
#define EXTRUDERS 2
#define SINGLENOZZLE
// #define MIXING_EXTRUDER
#define NUM_RUNOUT_SENSORS   2
#define FILAMENT_RUNOUT_SCRIPT "M600"
```

#### mixing extrude setup -> *"E stepper not specified"* error

``` C
#define EXTRUDERS 1
//#define SINGLENOZZLE
#define MIXING_EXTRUDER
#define NUM_RUNOUT_SENSORS   2
#define FILAMENT_RUNOUT_SCRIPT "M600"
```

### Steps to Reproduce

1. start a print
2. trigger the filament runout sensor

(or call `M600` via terminal)

**Expected behavior:**

Execute filament change for active extruder. Or even better, execute filament change for *triggering extruder*.

**Actual behavior:**

*"E stepper not specified"* error and print continues (without filament).

### Additional Information

I traced this down to different error handling (missing `Tx`) of the functions `get_target_extruder_from_command()` (used with dual extruders) and `get_target_e_stepper_from_command()` (used with mixing extruders). Is this intended?

Maybe it would be even better to be able to pass the triggering filament sensor to the runout script. e.g. `"M600 T%t"`

ellensp, 2020-10-24:
Is this on bugfix? As it needs tested on bugfix first to verify if it is still a issue or not.

KF1Z, 2020-10-27:
#13351

Vertabreak, 2020-10-27:
Last I tested the filament change was designated by the active vtool.

boelle, 2020-10-27:
> [BUG] Filament Change with Mixing Extruder #13351

that one got fixed and is working as it should

the-real-orca, 2020-10-28:
Bug  #19911 is related to the LCD menu. I’ll have a look at the fix.

The problem is that `Tx` is mandatory for `M600 `on mixing extruders. But there is no possibility to inject it into the runout script.

Yes, mixing extruders are tricky. Getting a single extrude from an active vtool may be even impossible for a true mix. IMHO it would be best to keep `M600 `as it is (with mandatory `Tx`) but add a functionality to be able to use the triggering sensor / extrude in the runout script.

e.g.
```C
#define FILAMENT_RUNOUT_SCRIPT "M600 T%t"
```

tommylux, 2020-11-19:
Sorry if this is the wrong place and I don't fully understand the Github workflow:
I have a A20T with runout sensors and it doesn't detect when the filament runs out. I understand this to be the bug relating to this issue. What is the status of it, has it been committed to the bugfix branch?
I am just working out how to compile the hex and youtube seems out of date.

Thanks

the-real-orca, 2020-11-21:
After some investigation, this bug seems to be a combination of 2 issues. Both are related to mixing extruder configurations.
1. Only the first filament sensor triggers the runout script for mixing extruders.
2. M600 requires an explicit tool parameter. The default to active extruder does not work for mixing extruders.

I’m working on a solution, but there is no official patch right now. You can check out my working branch if you would like to test it / give some feedback. 
https://github.com/the-real-orca/Marlin/tree/A20M_MixingTest
Make sure to adjust the configuration to your A20T (shouldn't be to different from the A20M I use for testing). Please keep in mind, that this is still work in progress.

tommylux, 2020-11-22:
#13351 suggests it has been fixed.. perhaps this means the runout sensor works when printing a single colour but if mixing with M163 then in doesn’t work? Unfortunately, I’m not familiar with the Git workflow. Perhaps I’m interested in your  branch, but what if there are major advancements in the master branch, does this make its way to forks?

the-real-orca, 2020-11-22:
#13351 has fixed the filament handling commands for mixing extruders. (i.e. use the correct extruder stepper). However there is still an issue with triggering the runout script (e.g. filament change) on mixing extruder setups. Currently the runout script is only triggered for the active extruder (virtual tool), however there is no dedicated (single) extrude for mixing extrudes. And the virtual tool number does not necessarily need to match an actual stepper. My approach is to use the triggering runout sensor as there should be a 1-to-1 match to the extruder stepper.

My branch is for development & testing only. After I have done more tests I’ll make a pull request to be merged with the official Marlin repository. However this can take some time. So if you want to help with testing in the meantime you can check out the development branch, do some tests and afterwards switch back to the official Marlin branch. 
I would be happy about any feedback if it works on other printers as well, especially with 3 mixing extruders.

Oversimplified workflow: *create development branch -> fix/update & test -> pull request to official bugfix branch -> more testing -> … next release*
If you are interested in a full description:
https://marlinfw.org/docs/development/getting_started_pull_requests.html

tommylux, 2020-11-22:
I’ll try and check it out. My point being if you make a development branch, it is only bugfixes at the time of the branch right? Also, my issue is, I tried compiling the bugfix branch with the configuration.h example file for A20T i found in configuration branch, then I had homing error when trying to print. So i’m little unsure what I did wrong. If you are able to fix this long running issue and commit to main branch, you’re a god in my eyes. I’ve got some experience in C, but developing firmware feels like a big undertaking.

the-real-orca, 2020-11-22:
Yes, the branch is for fixing one single issue.
The configuration of the A20T should be similar to the A20M, except for the additional extruder. Also make sure that you have selected the correct motherboard `BOARD_GT2560_V3_A20`, assuming you have a version 3 board.
If you need further assistance with the configuration, send me a PM (see profile).

Frontier314, 2020-11-24:
> I’m working on a solution, but there is no official patch right now. You can check out my working branch if you would like to test it / give some feedback.
> https://github.com/the-real-orca/Marlin/tree/A20M_MixingTest

I am in the process of upgrading my Creality CR-X to the latest Marlin 2.0.x-bugfix. Currently, I have working printer profiles for the base CR-X and today I have finished testing also dual filament runout sensor setup. I am also creating a different CR-X setup by using the BigTreeTech ZSYongMix hotend, this one is called CR-X Mix (again both with filament and no filament detection).

I can verify that your development code works fine on the CR-X Mix (I have cherry-picked your changes and updated my Marlin codebase). The following scenarios were tested, regarding filament detection (all files sliced with Cura 4.8.0 using a modified CR-X profile):

- Printing a file sliced for single extrusion (tested both extruders), with extruder selection from Cura drop-down menus. Code detects the filament sensor on the active extruder and ignores the sensor on the non-printing extruder.
- Printing a file sliced using each extruder, using mixing. Here, a mix is set, each time with one extruder at 100% mix. Again, the code detects the correct active extruder and ignores the one not printing.
- Printing a file sliced using Cura and post-processed with Geeetech's Mixing G-Code tool (gradient mix from 0% to 100%). The code monitors both sensors and the one that is triggered unloads the filament while the non-triggered does not change.

All these tests were done with the printer performing an advanced park of the nozzle.

I will keep using the code as I update the CR-X DWIN ExtUi code because it has some issues detecting clicks on the screen (has nothing to do with the core Marlin runout detection code, is only for CR-X). In any case, your changes make the runout code work as it should when one uses a mixing extruder set-up.

the-real-orca, 2020-11-25:
Thanks for your feedback and excessive testing. Good news to hear it’s also working on other printers.
I hope to find some time on the weekend to finalize my tests, code cleanup and submit a pull request.

tommylux, 2020-11-26:
How might I be able to load your changes for testing into my fork?

Thanks

Frontier314, 2020-11-27:
> 
> How might I be able to load your changes for testing into my fork? Thanks
> […](#)
> Sent from my iPad

the-real-orca's changes or mine?
I am not so familiar with Github but I can upload the whole source tree base somewhere.

tommylux, 2020-11-27:
I am using Verts branch because when I try the main / bugfix branch I get homing issues and I can’t work it out why. I tried coping all parameters from Verts and also tried example A20T configuration.h files, but still get homing issues.

Therefore, Vert must be doing something good for the build for my machine. How might I be able to insert your code in the Verts build for my testing?

I can see that Verts merges the bugfix branch into his fork, so if your code makes it up to the bugfix branch, it may come be merged into Verts.

Tom

KF1Z, 2020-12-10:
I can confirm that the-real-orca's  fix works for the geeetech mixing printers.
(thank you)

One thing to note however, ( I'm not sure where to report this)
For the A10T, and A20T printers, the configuration.h example file has pin 68 defined for FIL_RUNOUT3_PIN
This is incorrect.
The correct pin is 54

tommylux, 2021-01-11:
Greetings. Has a pull request been made yet for the runout sensor?

KF1Z, 2021-01-11:
yes.
https://github.com/MarlinFirmware/Marlin/pull/20327

tommylux, 2021-02-12:
I don't quite understand the pull request. I can see you want to merge 8 commits, have they all been accepted yet?

the-real-orca, 2021-02-13:
> I don't quite understand the pull request. I can see you want to merge 8 commits, have they all been accepted yet?

The pull request was originally submitted end November. Since then there have been some discussion, changes requests & updates. For the actual merge the 8 commits can be combined to 1 commit. Since December there where no new change requests and the pull request passes all tests, however it has not been merged to the bugfix branch yet. 
Maybe it got lost in the list of other pull requests.

alfvv12, 2021-03-20:
hello.
i used this configuration files with geeetech A20m and A20t.
no success.
the runout sensor doesn't work, no  message on display if no filament is available.

any suggestion?

thanks in advance
alfvv12

the-real-orca, 2021-03-23:
> i used this configuration files with geeetech A20m and A20t.
> no success.
> the runout sensor doesn't work, no message on display if no filament is available.

Are you using the latest firmware from the bugfix 2.0?

Another thing to keep in mind: The runout sensor only triggers during a running print.

alfvv12, 2021-03-24:
hi. thx for answer.
a) i use normal marlin 2.0 bugfix soft (tried also with yours but also no success)
b) printer A20M  (and A20T)

c) i changed in Configuration.h only this:

#define FILAMENT_RUNOUT_SCRIPT "M600 T%t"

// This defines the number of extruders
// :[0, 1, 2, 3, 4, 5, 6, 7, 8]
#define EXTRUDERS 1   
//#define SINGLENOZZLE    
#define MIXING_EXTRUDER    ****
#define NUM_RUNOUT_SENSORS   2    // 3 for a20T**

d) in old marlin 1.1.8  message "no filament" is shown on display, if no filament inserted. message is shown always, not only during printing process.
in marlini 2.0 no message is shown.

also if a print is running, no message if filament connector is disconnected.

e) so my question, what other changes are necessary?

thanks a lot for your reply in advance.

alf

alfvv12, 2021-04-01:
hi.

have anybody configuration files for geeetech A20m or A20T

with working filament runout?

that means, message "filament runout"  on the lcd display AND pausing printer.

only marlin 1.1.8 is working, with marlin 2.0 not. i tried often, no success. filament is running out and no stop and no message..

thx. for a link.

thinkyhead, 2021-04-02:
Please post a new bug report, since a PR has already been merged to address the original post.

closed 2021-04-02 by thinkyhead

localhorst24, 2021-04-02:
Hi.

M43 give me something confusing....

```
>>>M43
SENDING:M43
PIN:   0   Port: E0        RXD0                                   protected
.                          FIL_RUNOUT3_PIN                        protected
PIN:   1   Port: E1        TXD0                                   protected
PIN:   2   Port: E4        HEATER_2_PIN                           Output = 0    TIMER3B   PWM:     0    WGM: 1    COM3B: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   3   Port: E5        HEATER_1_PIN                           Output = 0    TIMER3C   PWM:     0    WGM: 1    COM3C: 0    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   4   Port: G5        FIL_RUNOUT_PIN                         protected
.                          HEATER_BED_PIN                         protected
PIN:   5   Port: E3        FIL_RUNOUT2_PIN                        Output = 1    TIMER3A   PWM:     0    WGM: 1    COM3A: 1    CS: 3    TCCR3A: 1    TCCR3B: 3    TIMSK3: 0
PIN:   6   Port: H3        <unused/unknown>                       Input  = 1                           TIMER4A   PWM:     0    WGM: 1    COM4A: 1    CS: 3    TCCR4A: 1    TCCR4B: 3    TIMSK4: 0
PIN:   7   Port: H4        FAN2_PIN                               protected
PIN:   8   Port: H5        FAN1_PIN                               protected
PIN:   9   Port: H6        FAN_PIN                                protected
PIN:  10   Port: B4        HEATER_0_PIN                           protected
PIN:  11   Port: B5        SERVO0_PIN                             Input  = 1    TIMER1A   PWM:   423    WGM: 4    COM1A: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode   compare interrupt enabled
PIN:  12   Port: B6        E0_ENABLE_PIN                          protected
PIN:  13   Port: B7        LED_PIN                                Input  = 1    TIMER0A   PWM:     0    WGM: 3    COM0A: 3    CS: 3    TCCR0A: 3    TCCR0B: 3    TIMSK0: 5   overflow interrupt enabled
 .                  TIMER1C is also tied to this pin                  TIMER1C   PWM:     0    WGM: 4    COM1C: 0    CS: 2    TCCR1A: 0    TCCR1B: 10    TIMSK1: 2   non-standard PWM mode
PIN:  14   Port: J1        <unused/unknown>                       Input  = 1
PIN:  15   Port: J0        <unused/unknown>                       Input  = 1
PIN:  16   Port: H1        BTN_EN1                                Input  = 0
PIN:  17   Port: H0        BTN_EN2                                Input  = 1
PIN:  18   Port: D3        BEEPER_PIN                             Output = 0
PIN:  19   Port: D2        BTN_ENC                                Input  = 1
PIN:  20   Port: D1        <unused/unknown>                       Input  = 1
PIN:  21   Port: D0        <unused/unknown>                       Output = 1
PIN:  22   Port: A0        <unused/unknown>                       Input  = 0
PIN:  23   Port: A1        Z_DIR_PIN                              protected
PIN:  24   Port: A2        X_MIN_PIN                              protected
.                          X_STOP_PIN                             protected
PIN:  25   Port: A3        Z_STEP_PIN                             protected
PIN:  26   Port: A4        <unused/unknown>                       Input  = 1
PIN:  27   Port: A5        Z_ENABLE_PIN                           protected
PIN:  28   Port: A6        Y_MIN_PIN                              protected
.                          Y_STOP_PIN                             protected
PIN:  29   Port: A7        Y_ENABLE_PIN                           protected
PIN:  30   Port: C7        Z_MIN_PIN                              protected
.                          Z_STOP_PIN                             protected
PIN:  31   Port: C6        Y_STEP_PIN                             protected
PIN:  32   Port: C5        <unused/unknown>                       Input  = 1
PIN:  33   Port: C4        Y_DIR_PIN                              protected
PIN:  34   Port: C3        <unused/unknown>                       Input  = 1
PIN:  35   Port: C2        X_ENABLE_PIN                           protected
PIN:  36   Port: C1        <unused/unknown>                       Output = 0
PIN:  37   Port: C0        X_STEP_PIN                             protected
PIN:  38   Port: D7        SD_DETECT_PIN                          Input  = 0
PIN:  39   Port: G2        X_DIR_PIN                              protected
PIN:  40   Port: G1        <unused/unknown>                       Input  = 1
PIN:  41   Port: G0        E2_ENABLE_PIN                          protected
PIN:  42   Port: L7        <unused/unknown>                       Input  = 1
PIN:  43   Port: L6        E2_STEP_PIN                            protected
PIN:  44   Port: L5        E0_DIR_PIN                             protected
PIN:  45   Port: L4        E2_DIR_PIN                             protected
PIN:  46   Port: L3        E0_STEP_PIN                            protected
PIN:  47   Port: L2        E1_DIR_PIN                             protected
PIN:  48   Port: L1        E1_ENABLE_PIN                          protected
PIN:  49   Port: L0        E1_STEP_PIN                            protected
PIN:  50   Port: B3        AVR_MISO_PIN                           Input  = 1
PIN:  51   Port: B2        AVR_MOSI_PIN                           Output = 1
PIN:  52   Port: B1        AVR_SCK_PIN                            Output = 0
PIN:  53   Port: B0        AVR_SS_PIN                             Output = 1
[... cut ...]
PIN:  65   Port: K3 (A11)  TEMP_0_PIN                             protected
PIN:  66   Port: K4 (A12)  <unused/unknown>   Analog in =     5   Input  = 0
PIN:  67   Port: K5 (A13)  <unused/unknown>   Analog in =     5   Input  = 0
PIN:  68   Port: K6 (A14)  <unused/unknown>   Analog in =   235   Input  = 0
PIN:  69   Port: K7 (A15)  POWER_LOSS_PIN                         Input  = 1
```

But my runout pins are defined different...
```
#define FIL_RUNOUT_PIN       66  //PK4 
#define FIL_RUNOUT2_PIN     67  //PK5
#define FIL_RUNOUT3_PIN     54  //PF0 - AT10T only

```