MarlinFirmware/Marlin #19883: [BUG] CONTROLLERFAN_SPEED_ACTIVE can’t be modified when activating "LASER FEATURE"

I3megaS, 2020-10-24:
### Bug Description

I usually use CONTROLLERFAN_SPEED_ACTIVE at 190 value.
When I bought a laser and I activate LASER FEATURE. After a time I have noticed that the controller fan was not working as it should, and I have proved that the CONTROLLERFAN_SPEED_ACTIVE needs to be at 255 to work correctly. If I use any other value the controller fan stops, but I have checked that the pin is activated to a very low voltage value.

### My Configurations

[Config.Files.zip](https://github.com/MarlinFirmware/Marlin/files/5433747/Config.Files.zip)

### Steps to Reproduce

1. Change CONTROLLERFAN_SPEED_ACTIVE at 190 value
2. Test the controller fan by moving any axis. It works correct.
3. Now, activate LASER FEATURE
4. Test the controller fan again by moving any axis. It doesn't work.
5. Change  CONTROLLERFAN_SPEED_ACTIVE to 255 value. The controller fan starts turning.

**Expected behavior:**
With or without LASER FEATURE function activated, the controller fan should turn to the specified CONTROLLERFAN_SPEED_ACTIVE value.

**Actual behavior:**
When activating LASER FEATURE the controller fan only works if the CONTROLLERFAN_SPEED_ACTIVE value is 255.

#### Additional Information

The printer is an Anycubic I3 Mega-S (Trigorilla board) with TMC2209 full Uart, a laser with PWM input of 20kHz and powered with an external power supply, and Marlin FW 2.0.7.2
I have also do the same test over Marlin 2.0.7.1 and 2.0.6.1 with the same results.

shitcreek, 2020-10-31:
Quick question. Are your fan and laser assigned to different pins?

I3megaS, 2020-10-31:
yes, I am using different pins.
To control the laser I am using the servo pin D6 (PWM 5V) on Trigorilla board.

boelle, 2020-11-22:
> Marlin FW 2.0.7.2

> Marlin 2.0.7.1 and 2.0.6.1

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I3megaS, 2020-11-22:
I tested the firmware of your link and I get more or less the same results as before.
I have noticed that the fan is being driven by the negative terminal, so a lower duty cycle gives more voltage/current at the output for the fan. 

I have made some test with M710 instruction with the following results:

      M710 S150 - PWM duty cycle of 80% - approx 4V on terminals (not enough torque to start)
      M710 S190 - PWM duty cycle of 75% - approx 4,9V on terminals (not enough torque to start)
      M710 S220 - PWM duty cycle of 70% - approx 5,3V on terminals (not enough torque to start)
      M710 S254 - PWM duty cycle of 68% - approx 5,8V on terminals (enough torque to start)
      M710 S255 - PWM duty cycle of 0% - approx 12,8V on terminals (runs full power)

closed 2020-12-30 by github-actions[bot]