MarlinFirmware/Marlin #19884: [BUG] Failed to compile with password feature

Speaka, 2020-10-24:
### Bug Description
I rebased my working branch and failed to build. So I tried with a clean branch.
This will also bring up the same result and wont build.
Lmk if you need more Infos.

### My Configurations

Stock config with this two features enabled.
Building in Platformio on two different Systems. (Win10)

### Steps to Reproduce

Latest Bugfix with enabled:
_#define EEPROM_SETTINGS
#define PASSWORD_FEATURE_

**Expected behavior:**
Building successfully
**Actual behavior:**
Failing to build with following error:

`Marlin\src\feature\password\password.cpp:49:37: error: no 'void Password::authentication_check()' member function declared in class 'Password'
 void Password::authentication_check() {
                                      ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'

*** [.pio\build\mega2560\src\src\feature\password\password.cpp.o] Error 1`

ellensp, 2020-10-25:
you need HAS_LCD_MENU to be valid  to define this authentication_check function.
Ie you need a LCD display

ellensp, 2020-10-25:
In password.h is the following code

```
  #if HAS_LCD_MENU
    static void access_menu_password();
    static void authentication_check();
```
Change to 
```
  static void authentication_check();
  #if HAS_LCD_MENU
    static void access_menu_password();
```
Please confirm this works and you can unlock the printer.

ellensp, 2020-10-25:
PR created

ellensp, 2020-10-25:
Merged, closing.

closed 2020-10-25 by ellensp

Speaka, 2020-10-25:
Thanks for the quick support! All working fine now!