MarlinFirmware/Marlin #19889: [BUG] Z_STEPPER_AUTO_ALIGN Saves Alignment Points to EEPROM

NateroniPizza, 2020-10-25:
### Bug Description

When using Z_STEPPER_AUTO_ALIGN (G34), and alignment probe points are not defined, it will save the automatically-calculated points to EEPROM. If we modify the PROBING_MARGIN value, it will not recalculate until EEPROM is re-initialized.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/5434199/Configuration_Files.zip)

### Steps to Reproduce

1. Set a small value for PROBING_MARGIN (in my case, this was 5)
2. Ensure "#define Z_STEPPER_ALIGN_XY" is commented out.
3. Upload config to printer and run a G34.
4. Increase the value of PROBING_MARGIN (in my case, I increased this to 15)
5. Re-upload and re-run G34. You will receive the following message:
```
G34 Iteration: 1
Probing failed
G34 aborted.
```
(upon looking at verbose logs, it was still trying to probe at the original probing points)
6. Re-initialize EEPROM and re-test. It should now succeed, and the PROBING_MARGIN value.

**Expected behavior:** Re-calculate auto-align probe points with each run of G34

**Actual behavior:** It saves auto-align probe points to EEPROM, and does not re-calculate when values are changed in the configuration.

#### Additional Information

sjasonsmith in the Discord suggested submitting a bug report for this issue, as he has seen this issue come up twice in the past week.

sjasonsmith, 2020-10-25:
Ideally they would only be stored to EEPROM if they have been modified through M422, else use the defaults from the header.

This example was due to a PROBING_MARGIN change, which is a header-file change. We would also need to consider how this behaves if M851 is used to alter a probe offset.

sjasonsmith, 2020-10-26:
I prototyped a solution for this. It is a little sloppy, but I believe works if you want to try it. You will need to re-initialize your EEPROM after updating to it, to remove any previously stored G34 values.

This does the following:
- Does not store auto-generated G34 points to EEPROM
- Does store G34 points if you modify then using M422
- M422 and G34 immediately reflect probe offset changes, whether through M851 or an LCD menu.

https://github.com/sjasonsmith/Marlin/tree/InWork/Bug19889_G34_DoNotSaveAutoPoints

sjasonsmith, 2020-10-31:
@NateroniPizza did you have a chance to try the prototype I posted above?

NateroniPizza, 2020-10-31:
> @NateroniPizza did you have a chance to try the prototype I posted above?

Sorry, I missed the previous status update. I assume I would just download the firmware from that location, drop in my configuration.h and configuration_adv.h, and re-test? Are there any changes that need to be made to my configuration.h and configuration_adv.h for it to work? (created using 2.0.7.2 base configuration files).

sjasonsmith, 2020-10-31:
No configuration changes needed. Your current conditions if s should work wIt’s it if they work in the bugfix-2.0.x branch.

NateroniPizza, 2020-11-01:
I'm running into the following error when attempting to compile with my config:
[image]

With the stock config (just modifying the board), it succeeds.

Looking through my configs and the stock configs, I do not see anything about manual feedrate. Any idea what my configs are missing?

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5470450/configs.zip)

ellensp, 2020-11-01:
You haven't updated your configs from a previous version correctly 

In Configuration_adv.h
`#if EITHER(ULTIPANEL, EXTENSIBLE_UI)`  should be `#if EITHER(IS_ULTIPANEL, EXTENSIBLE_UI)`

ellensp, 2020-11-01:
also   `#if ENABLED(ULTIPANEL)` should be   `#if IS_ULTIPANEL
`

NateroniPizza, 2020-11-01:
> also `#if ENABLED(ULTIPANEL)` should be `#if IS_ULTIPANEL `

Changed, thank you. Compiled successfully, will load onto tomorrow.

thinkyhead, 2020-11-01:
@NateroniPizza — You need to update your `Configuration_adv.h` replacing `ULTIPANEL` with `IS_ULTIPANEL`…
```cpp
#if EITHER(IS_ULTIPANEL, EXTENSIBLE_UI)
  #define MANUAL_FEEDRATE { 50*60, 50*60, 4*60, 2*60 } // (mm/min) Feedrates for manual moves along X, Y, Z, E from panel
  #define SHORT_MANUAL_Z_MOVE 0.025 // (mm) Smallest manual Z move (< 0.1mm)
  #if IS_ULTIPANEL
    . . .
```
You might want to compare with the latest configuration files and apply any other recent updates as well.

NateroniPizza, 2020-11-02:
Hmm... No dice. Ran through the following:
1. Compiled and uploaded the bugfix firmware. Confirmed the startup screen does show bugfix-2.0.x (previously showed 2.0.7.2). Value of PROBING_MARGIN was 25.
2. Ran m502, then m500 to force a clear of eeprom.
3. Ran a home then g34.
4. Increased the value of PROBING_MARGIN to 50, then recompiled and uploaded.
5. Ran a home and a g34.

Result:
`SENDING:G34
G34 Iteration: 1
Probing failed
G34 aborted.`

Enabling verbose logging, it appears it is still trying to go to 25:
```
Probing X25.00 Y100.00
>>> Probe::probe_at_point  X127.00 Y42.00 Z12.65
...(25.00, 100.00, raise, 0, probe_relative)
  current_position= X127.00 Y42.00 Z12.65 :
Position Not Reachable
```

sjasonsmith, 2020-11-02:
You have to use the development branch in the link I provided. Not bug fix.

NateroniPizza, 2020-11-02:
> You have to use the development branch in the link I provided. Not bug fix.

That's the one I downloaded. I don't even know where I'd download the bugfix version from.

sjasonsmith, 2020-11-02:
I will have to look at it later. Do you have anything that would issue M422 commands to set points? Setting points that way will cause it to save them and not re-calculate.

Can you send M422 with no arguments before and after you change the probe margin? If my change is working that should show re-calculated points each time.

NateroniPizza, 2020-11-02:
> I will have to look at it later. Do you have anything that would issue M422 commands to set points? Setting points that way will cause it to save them and not re-calculate.
>
> Can you send M422 with [...]

I was just running G34 - what sorts of things would issue an M422?
Ended up starting a relatively short print - will try the M422 after that complete.

NateroniPizza, 2020-11-05:
> I will have to look at it later. Do you have anything that would issue M422 commands to set points? Setting points that way will cause it to save them and not re-calculate.
>
> Can you send M422 with [...]

Interesting... I must've been doing something wrong. I just went through it again, and all appears to be working.

sjasonsmith, 2020-11-05:
Thanks for letting me know. I hadn’t had a chance to get back to it. I don’t think this branch is in a good state yet, I’ll need to look back at it and decide what a final form should look like.

NateroniPizza, 2020-11-05:
> Thanks for letting me know. I hadn’t had a chance to get back to it. I don’t think this branch is in a good state yet, I’ll need to look back at it and decide what a final form should look like.

Sounds good. This was discovered and tested in a friend's 3D printer that I was fixing up - it was returned to them this evening, so I no longer have access to a printer running Marlin for further testing.

InsanityAutomation, 2020-11-10:
Slightly different path than what I had originally though of, which was forcing recalc on M851 but same goal achieved.

sjasonsmith, 2020-11-10:
@InsanityAutomation forcing recalc still has its place in my approach, if you want to clear customized points without completely resetting your EEPROM.

My hope was that for a typical user they would use automatically selected points and never thing about it if they happened to change values impacting the probable area.

I’d appreciate any feedback on the approach. You shouldn’t pay too much attention to the actual code as I need to rework it beyond the current “thrown together proof of concept” state.

sjasonsmith, 2020-11-10:
I read your reply again. I actually was forcing recalc on M851, but encountered a problem when setting from the LCD. The LCD doesn’t work through M851, it sets variables in-place. Rather than rework the menus and possibly find other similar issues I chose to just never store the values if they were automatic values. It only runs once per print at most, so I didn’t see any issue with that.

InsanityAutomation, 2020-11-10:
For the menu i probably would have used the callback functions and just issued an inject command for G34R. I only glanced quickly. My head is finally unfogged enough from the day job to think about some of this again, not going to refog it deep diving lol

closed 2021-01-17 by github-actions[bot]