MarlinFirmware/Marlin #19891: [BUG] (problems with russian localization, mksui)

Malderin, 2020-10-25:
Board: mks robin nano v2
My configs: [conf.zip](https://github.com/MarlinFirmware/Marlin/files/5434277/conf.zip)
Marlin: last bugfix-2.0.x

After committing this bug report https://github.com/MarlinFirmware/Marlin/issues/19872 new problems found.
1. the problem with the display of the russian translation in the dialog box is not fixed
See the first message of this report: https://github.com/MarlinFirmware/Marlin/issues/19872
see this video: https://yadi.sk/i/b4iCmifkI9RTkQ
2. New dialog box issue. When you start to print gcod file from a memory card, when you press "cancel", a black screen
see video: https://yadi.sk/d/CYo6AfHiZ06nEg
3. Menu display problem. After moving through the menu, some of the icons crumble.
see video: https://yadi.sk/i/g4cdXzrOD3JD9A

Malderin, 2020-10-25:
> 2. New dialog box issue. When you start to print gcod file from a memory card, when you press "cancel", a black screen
>    see video: https://yadi.sk/d/CYo6AfHiZ06nEg
> 3. Menu display problem. After moving through the menu, some of the icons crumble.
>    see video: https://yadi.sk/i/g4cdXzrOD3JD9A

These problems appeared after this commit:
https://github.com/MarlinFirmware/Marlin/commit/853d01de683ab3efcbb15ef95db869e279053ee0

> 3. Menu display problem. After moving through the menu, some of the icons crumble.
>    see video: https://yadi.sk/i/g4cdXzrOD3JD9A

I think it has to do with the header update. For example, I write 1 in the draw_tool.cpp file, i.e. it turns out like this:
void lv_draw_tool (void) {
   scr = lv_screen_create (TOOL_UI);
   lv_refr_now (lv_refr_get_disp_refreshing ());
After that, the display of icons in the "Home" menu is normal. 
If you do the same in the menu "Tool" (draw_tool.cpp) and "Move" (draw_move_motor.cpp), the display problem disappears.
I'm not sure if such actions are correct.

rhapsodyv, 2020-10-26:
Sorry about that issues. We did a strong refactoring in the LVGL and there're more than 30 screens to test. 

I think I fixed the remaining issues here: #19904

Thanks!

Malderin, 2020-10-26:
> Sorry about that issues. We did a strong refactoring in the LVGL and there're more than 30 screens to test.
> 
> I think I fixed the remaining issues here: #19904
> 
> Thanks!

I checked, all 3 issues are fixed, but I noticed new problems.
1. When trying to enter the "Tool / Move" menu the printer restarts;
2. In the "settings / config" menu, all parameters in all menu items have disappeared.
For example, the "settings / config / machine settings / acceleration settings" menu looks like this:
[image]
It is also impossible to enter data into these fields from the keyboard.

rhapsodyv, 2020-10-26:
Ok, I will check that.

rhapsodyv, 2020-10-26:
I did a commit now. Can you test it?

I had no issue with the Tool/Move screen. It works fine for me.

Malderin, 2020-10-26:
> I did a commit now. Can you test it?
> 
> I had no issue with the Tool/Move screen. It works fine for me.

what is the commit number?

rhapsodyv, 2020-10-26:
I just pushed to the same branch: https://github.com/MarlinFirmware/Marlin/pull/19904

Malderin, 2020-10-26:
> I just pushed to the same branch: #19904

I checked, now everything is fine!

PR #19904 (2020-10-27): LVGL follow up (2)
### Description

Fix correct stacking for printing screen, missing label style and draw order.

### Benefits

Fix #19891
closed 2020-10-26 by Malderin