MarlinFirmware/Marlin #19894: [BUG] Weird temperature spikes with newest stable version of Marlin 2.0.7.2

szymonrychu, 2020-10-25:
### Bug Description

Hello!
I would like to report a weird problem that occurs on a my DUE based environment after upgrading to newest 2.0.7.2 version of Marlin.
The problem IS NOT hardware, occurs only after flashing newest version of the firmware.
My hardware consist of DUE3DOM_LE (DUE3DOM_MINI based board), TMC2130 drivers on all axises, dual Z screw independent Z steering, UBL enabled with BLTouch and with E3D pt100 temperature sensor with their amplifier.

What I observed is that during prints Marlin is reporting weird temperature spikes pointing "down" and equalling always 0degC.
Such spikes doesn't trigger overheat protection safeties.
Again- those are not happening with older versions of marlin.

I suspect that there are two separate loops- one responsible for printing out stuff to terminal and the other responsible for reading out temps. Sometimes those two get out of sync somehow, therefore 0 gets printed out to terminal.
Here is the code [that was working for me before](https://github.com/szymonrychu/Marlin/tree/DUE3DOM_LE) with last common git SHA: `1ac629f919a6312f1ff5c8736e3cc8bea3f2c2b5`.

### My Configurations

I forked Marlin to my repo and work on a daily basis with GIT so instead of uploading zips I will attach a link to a compare page between what I have and original 2.0.7.2 revision of a repo.
[https://github.com/MarlinFirmware/Marlin/compare/2.0.7.2...szymonrychu:DUE3DOM_LE_develop](https://github.com/MarlinFirmware/Marlin/compare/2.0.7.2...szymonrychu:DUE3DOM_LE_develop)
Besides having Configs the delta also includes board definition.

### Steps to Reproduce

Apply patch from a link, flash to DUE, start print and observe temperatures I guess? Hard to tell what might be causing this.

szymonrychu, 2020-10-25:
Screenshot showing off severity of the problem: [https://imgur.com/a/nwptVr9](https://imgur.com/a/nwptVr9)

sjasonsmith, 2020-10-26:
@szymonrychu you say this is a recently introduced problem. Could you try to isolate which commit caused it?

szymonrychu, 2020-10-26:
Out of 2514 commits of difference between latest known good revision up to the first commit that might cause this problem?
I mean- sure I can... But I don't know when I would have enough time to do this- assuming process like this:
1. Rebase
2. Rebuild
3. Flash the new code
4. Test print
would be fully automated (which might or might not be) and would take roughly speaking 5mins in total, it's still 12500min, which more than 8days of continous testing.

At the other hand if something is tagged as a stable release and if there is a documentation stating, that it's supporting certain boards and processors, I would expect that it's sufficiently tested.
In the end we are talking even about potential fire-hazard if it's not, right?

But in the end I can give you some direction I guess? Something I've tested yesterday and I didn't put in the comments yet.

It turns out that last known working revision is showing the same symptoms when I enable `S_CURVE_ACCELERATION`.
At the same time it really doesn't make any difference for the code from `2.0.7.2` version if I enable it or not.

szymonrychu, 2020-10-26:
Edit:
Honestly speaking- not recently... Last known common commit is https://github.com/szymonrychu/Marlin/commit/1ac629f919a6312f1ff5c8736e3cc8bea3f2c2b5 and is dated to `Oct 5, 2019`.

rhapsodyv, 2020-11-30:
Can you share the serial output?

sjasonsmith, 2020-12-20:
@szymonrychu are you still having this problem? Did you see @rhapsodyv's request for the serial output when this is occurring? You should be able to see it in the OctoPrint terminal. We want to see what the temperate reporting looks like in text form.

sjasonsmith, 2020-12-29:
No response after 9 days. Closing.

closed 2020-12-29 by sjasonsmith