MarlinFirmware/Marlin #19895: Marlin 2.0.7.2 failed to compile with the feature of "controller fan" for arduino DUE

hkckko, 2020-10-25:
**Marlin 2.0.7.2
Arduino IDE 1.8.13 running under ubuntu OS
Target: Arduino Due (Programming Port)**
**When enable the** "**USE_CONTROLLER FAN**" **in** "**Configuration_adv.h**"
 
_#define USE_CONTROLLER_FAN_
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN 11        
  //#define CONTROLLER_FAN_USE_Z_ONLY    // With this option only the Z axis is considered
  //#define CONTROLLER_FAN_IGNORE_Z      // Ignore Z stepper. Useful when stepper timeout is disabled.
  #define CONTROLLERFAN_SPEED_MIN      0 // (0-255) Minimum speed. (If set below this value the fan is turned off.)
  #define CONTROLLERFAN_SPEED_ACTIVE 255 // (0-255) Active speed, used when any motor is enabled
  #define CONTROLLERFAN_SPEED_IDLE     0 // (0-255) Idle speed, used when motors are disabled
  #define CONTROLLERFAN_IDLE_TIME     10 // (seconds) Extra time to keep the fan running after disabling motors
  //#define CONTROLLER_FAN_EDITABLE      // Enable M710 configurable settings
  #if ENABLED(CONTROLLER_FAN_EDITABLE)
  #define CONTROLLER_FAN_MENU          // Enable the Controller Fan submenu
  #endif
  #endif_

**The following compiling errors were observed**:

_In file included from sketch/src/MarlinCore.cpp:209:0:
sketch/src/feature/controllerfan.h:61:67: error: non-constant in-class initialization invalid for static member 'ControllerFan::settings'
         static const controllerFan_settings_t constexpr &settings = controllerFan_defaults;
                                                                   ^
sketch/src/feature/controllerfan.h:61:67: error: (an out of class initialization is required)
sketch/src/feature/controllerfan.h:61:67: error: 'ControllerFan::settings' cannot be initialized by a non-constant expression when being declared
exit status 1
Error compiling for board Arduino Due (Programming Port)._

boelle, 2020-10-25:
does this also happen if you use vscode+platform.io ?
(this is the prefered choice as arduino ide is filled with bugs that only arduino people can fix - and they dont)

i use a re-arm and not the due but i dont have any compile errors

ellensp, 2020-10-25:
What DUE based motherboard, there are 28 possibilities...

ellensp, 2020-10-25:
I guessed a motherboard, added a CONTROLLER_FAN_PIN and can replicate the error in arduinio ide. 
PlatformIO does not error. and takes about 1/3 or the time to compile.

ellensp, 2020-10-25:
Enabling CONTROLLER_FAN_EDITABLE bypasses the the issue

ellensp, 2020-10-25:
Looks like the following line in Marlin/src/feature/controllerfan.h really shouldn't have the &. 
`      static const controllerFan_settings_t constexpr &settings = controllerFan_defaults;`

hkckko, 2020-10-25:
Thank your for boelle's and ellensp's advice.

It is success to compile with PlatformIO under windows environment. 
I think the problem shall be caused by the Arduino IDE.

shitcreek, 2020-10-29:
The & operator seems to cause issue for STM32 in general. Such as:
https://github.com/MarlinFirmware/Marlin/issues/19431#issuecomment-717607577

sjasonsmith, 2020-11-07:
A pull request is posted to fix this.

sjasonsmith, 2020-11-15:
My changes have been merged. Closing as presumed fixed.

closed 2020-11-15 by sjasonsmith