MarlinFirmware/Marlin #19896: [BUG] Weird Output after G35

Speaka, 2020-10-25:
### Bug Description
Performing G35 responsing weird characters via Serial (Octoprint). It says how to turn but the point name is cryptic.
_Recv: Turn ¦¬¬¬¬¬¬¬¬¬¬¬¬¬¬..._
Outputlog [here](https://controlc.com/6260812f)
### My Configurations

See my fork here: https://github.com/Speaka/Marlin/tree/hypercube_work/Marlin
Based on latest Bugfix.

### Steps to Reproduce
G35
Wait for all probe steps to finish
See serial response spammend with unreadable characters

**Expected behavior:** [What you expect to happen]
Clean Output
**Actual behavior:** [What actually happens]
Weird Output
#### Additional Information

It seems theres something Wrong with the TRAMMING_POINT_NAME_1 to 4.

sjasonsmith, 2020-10-25:
Likely related or duplicate of #19555 or #19473.

#19473 has code attached they claim fixes the problem, if it is the same. You could try that out and see if it works.

Speaka, 2020-10-25:
Thanks for the quick response!
It was indeed the same problem.
[This Commit](https://github.com/RealDeuce/Marlin/commit/486437ed5f6bf1644fc5802a1caa2a73a4b735ac) did the trick.
It seems I am not the only one with this problem. Is there any reason why this isn't implemented or worked on? Or just not happend yet?
I can't oversee the whole serial communication and different platform thing. Otherwise i would look after it.

sjasonsmith, 2020-10-25:
With most work being done by volunteers it depends on someone stepping forward to do it.

sjasonsmith, 2020-10-25:
A few factors that I suspect have led to this issue being ignored:

- Most active contributors use 32-bit boards, and this issue only impacts 8-bit
- More experiences users already have their habits set, so this assistant is more likely used by newer users.
- People are busy, not just with Marlin. While it would be nice to tell people “you must fix this!”, that not really how semi-anonymous volunteers work!

Speaka, 2020-10-25:
Don't get me wrong! I am totally aware how things work based on volunteers.
Asking why it wasn't done aimed for a reason like incompatibilities, not someone not feeling like doing it or having time to.
Youve got valid points there. I have an 8bit System and pretty wasted bed so I would be predestined to get this to work somehow.
I'll try to get this straight and file a PR in he next days.
Needs testing on other Systems then to prevent converting the bug.

sjasonsmith, 2020-10-25:
@Speaka thanks, that would be great!

sjasonsmith, 2020-10-25:
When you have it ready, just post the PR. You can ask for somebody to try it on a 32-but board from there.

sjasonsmith, 2020-10-26:
I'm going to go ahead and close this as a duplicate of the other. Any commentary leading up to a pull request can happen over there.

sjasonsmith, 2020-10-26:
Duplicate of #19473

closed 2020-10-26 by sjasonsmith

Speaka, 2020-10-27:
PR created