MarlinFirmware/Marlin #19897: [BUG] Display will turn black when increasing BAUDRATE for Ender 3 v2

foxylion, 2020-10-25:
### Bug Description

When I increase the BAUDRATE from the default `115200` to  `250000` the screen of the printer will stay black after powering on the printer. It is usable via USB.

I would expect that changing the `BAUDRATE` property should not have any effects on the display.

### My Configurations

See attached the used configurations (based on the Ender 3 v2 example configurations.
Only thing modified is `#define BAUDRATE 250000`.

[Ender-3 V2.zip](https://github.com/MarlinFirmware/Marlin/files/5435000/Ender-3.V2.zip)

### Steps to Reproduce

1. Buy an Ender 3 v2 ;-)
2. Copy the configuration files from [here](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2) into the Marin project to `./Marlin`.
3. Build the project
4. Flash the firmware
  - Copy the firmware onto the SD card
  - Plug-in the SD card into the printer, power it on
  - Wait for firmware to be flashed
  - Turn of the power of the printer
  - Flash the screen with the `DWIN_SET` files
5. Check that everything works as expected ✔️ 
6. Change the `BAUDRATE` property in `Configuration.h` to `250000`.
7. Build the project again
8. Repeat the flashing-process (used another filename for the firmware file)
9. Screen does no longer turn on after flashing :x:

When rolling back the the firmware built in step 3 everything works fine again.

**Expected behavior:** The screen should turn on, even when using `#define BAUDRATE 250000`

**Actual behavior:** The screen doesn't turn on.

#### Additional Information

I also tried to flash the printer/screen multiple times, because that was mentioned in some Facebook group, but did not help.

ellensp, 2020-10-26:
The DWIN_CREALITY_LCD is a serial device! Marlin by default uses a global baudrate. 
So yes, changing the baudrate breaks your lcd, as expected. 
NOT A BUG

closed 2020-10-26 by ellensp

sjasonsmith, 2020-10-26:
You may find a workaround documented here: https://github.com/MarlinFirmware/Marlin/issues/18231

ellensp, 2020-10-26:
Or customize the code a little
Find  `MYSERIAL1.begin(BAUDRATE); `  in Marlin/src/MarlinCore.cpp
Change to `MYSERIAL1.begin(115200);`   makes your display always 115200 independent of BAUDRATE

foxylion, 2020-10-26:
Thank you @ellensp and @sjasonsmith. Sorry for reporting this as a bug. But thank you for your help!